RobotLocomotion/drake #9428: Speed up contact Jacobians computation in MBP

amcastro-tri, 2018-09-11:
## Description

From #8942 we see that the computation of the contact Jacobians is one of the most time consuming operations during simulation with contact.

A more careful profiling of #8942 allows us to write down a more detailed budget for the normal and tangential Jacobian computations.

## Reproduction

Instruction on how to reproduce these profiling results are documented in #8942.

## Profiling Results

In a computation with 20,000 time steps, most of our computation is performed in `MultibodyPlant::DoCalcDiscreteVariableUpdates()`, 91% to be more precise (it'd be very nice to understand why we have such a high 10% overhead for this simple case where all computation should happen in this method. Ideally we'd see this method taking 100% of the cost).

The **relative** breakdown of `MultibodyPlant::DoCalcDiscreteVariableUpdates()` is:
- CalcTangentVelocitiesJacobian(): 26%
- CalcNormalSeparationVelocitiesJacobian(): 25%
- CalcPointPairPenetrations(): 22%
- CalcMassMatrixViaInverseDynamics(): 13%
- SolveUsingSubStepping(): 8%

### Some Remarks

1. `CalcTangentVelocitiesJacobian()` and `CalcNormalSeparationVelocitiesJacobian()` are coupled in that they both call `MBT::CalcPointsGeometricJacobianExpresedInWorld()`. **There is low hanging fruit here in that we can combine both methods into one to get rid of a 25% computation cost.**
2. All MBT methods here for Jacobian and mass matrix are at least re-computing the position kinematics. **A lot** of the extra cost can be reduced if we at least implement caching for the position kinematics.

## Action Items

- [ ] Factor `CalcTangentVelocitiesJacobian()` and `CalcNormalSeparationVelocitiesJacobian()` into a single method to re-use common computations.
- [ ] Implement position kinematics caching in MBP/MBT.
- [ ] Profile the example again after these improvements are introduced to find out next bottleneck in the chain.

amcastro-tri, 2018-09-11:
cc'ing @rcory, @sherm1,  @SeanCurtis-TRI and @DamrongGuoy .

@DamrongGuoy and @SeanCurtis-TRI, notice that a more careful profiling will be needed in `CalcPointPairPenetrations()` and I'm sure we could do better.

SeanCurtis-TRI, 2018-09-11:
Quick question (now that github is finally working for me). Why merge the two functions? Can't their common dependency be a cache entry? I'm assuming there was a good reason to have separate functions in the first place. A cache entry would allow you to maintain that separation and avoid the duplicate calculations.

amcastro-tri, 2018-09-11:
Quick answer, we don't have caching in MBP. The two methods are small enough that merging them together won't cause any trouble. They are private anyways.
First spiral to caching implementation is tracked in #9429.

SeanCurtis-TRI, 2018-09-11:
Well, *technically*, you *do* have caching in MBP. :) You're just not explicitly using. But I take your point.

amcastro-tri, 2018-09-12:
#9433 merged.
New profiling [here](https://github.com/RobotLocomotion/drake/issues/8942#issuecomment-420656038) in #8942.

amcastro-tri, 2018-09-13:
## Update on the Jacobian computations breakdown.

#9433 Merges the contact Jacobians computations into a single method `CalcNormalAndTangentContactJacobians()` in order to re-use computations.

The central component of `CalcNormalAndTangentContactJacobians()` is `MBT::CalcPointsGeometricJacobianExpressedInWorld()`. Since for this case there are 16 contact pair pints, we have 32 calls per time step to this MBT method. That is, `MBT::CalcPointsGeometricJacobianExpressedInWorld()` **is responsible for that 26%** of the total cost.

The profiling of `MBT::CalcPointsGeometricJacobianExpressedInWorld()` breaks down into two major components (**relative to this method**, not the total):
- `MBT::CalcAcrossNodeGeometricJacobianExpressedInWorld()`: 47%. Called [here](https://github.com/RobotLocomotion/drake/blob/master/multibody/multibody_tree/multibody_tree.cc#L927).
- `MBT::CalcPositionKinematicsCache()`: 31%. Called [here](https://github.com/RobotLocomotion/drake/blob/master/multibody/multibody_tree/multibody_tree.cc#L923).

Both of these two costs are a function of `q` only and therefore can be cached (there are TODOs in the code). If we do cache these methods the cost of the Jacobian cmputation can be made negligible since **for this case it'd result in about 1/30th of the current cost**.

@sherm1, these are the two major bottlenecks in the multibody code regarding speed. Caching `H_PB_W` and the position kinematics will have a major effect.

sherm1, 2018-09-13:
Awesome -- I'll work on getting these cached. Are there instructions somewhere for how you measured these results, Alejandro? I'd like to perform the identical measurement with and without caching.

amcastro-tri, 2018-09-13:
Excellent!. Yes, see at the top of this issue. Instructions were provided in [8942](https://github.com/RobotLocomotion/drake/issues/8942).

amcastro-tri, 2018-11-29:
Compared to the profiling from Sep 13 2018, `MBT::CalcPointsGeometricJacobianExpressedInWorld()` is responsible for that 12% of the total cost (compared to the 26% from Sep 13).

The breakdown from Sep 13 is out of date and the breakdown reduces to a single call  (**relative to this method**, not the total)::
- CalcFrameJacobianExpressedInWorld(): 91.64%.

That is, the cost to previous calls to `MBT::CalcAcrossNodeGeometricJacobianExpressedInWorld()` and `MBT::CalcPositionKinematicsCache()` was made negligible when introducing caching.

amcastro-tri, 2018-11-29:
The action items for this issue were completed. Full profiling results for the simple gripper example are provided in #8942 and above for particulars relevant to Jacobian computation.

closed 2018-11-29 by amcastro-tri