RobotLocomotion/drake #9443: geometry: support Convex Shape

DamrongGuoy, 2018-09-12:
Add `Convex` to the list of supported `Shape` (shape_specification.h) in `geometry`.  Currently we support `Sphere`, `Cylinder`, `Box`, and `HalfSpace` with limited support for `Mesh`.

sherm1, 2018-09-12:
See also #7656

DamrongGuoy, 2018-09-12:
This is my plan.
1. New class `geometry::Convex` as a `final` subclass of `Shape` (shape_specification.h)

2. Add `virtual ShapeReifier::ImplementGeometry(const Convex& convex,...)=0;` interface (shape_specification.h)

3. Define ImplementGeometry(...Convex&...) in these three subclasses of ShapeReifier:-
3.1 geometry::anonymous::ShapeToLcm (geometry_visualization.cc)
3.2 geometry::internal::ProximityEngine<T>::Impl (proximity_engine.cc)
3.3 geometry::anonymous::ReifierTest (shape_specification_test.cc)

I'm still studying how other `Shape` define 3.1-3.3 and will update again.

@sherm1 and @SeanCurtis-TRI, please let me know if there are more steps to add.

DamrongGuoy, 2018-09-14:
Thanks to @SeanCurtis-TRI , I have a preliminary version of `geometry::Convex` modeled after `Mesh` like this (here stripped down for readability):
```
class Convex : Shape {
public:
  Convex (const std::string& filename, double scale=1.0);
private:
  std::string filename_;
  double scale_;
};
```
As you can see, my `drake::geometry::Convex` only stores the file name.  However, for task 3.2 `ImplementGeometry(Convex)` for `ProximityEngine`, I will convert it to `fcl::Convex` which requires these data:
```
fcl::Convex (int num_vertices, Vector3<S>* vertices, int num_faces, int* faces);
```
Any advice how to do that, please, @sherm1 and @SeanCurtis-TRI ?  I'm afraid I cannot model it after `ImplementGeometry(Mesh)`, which is creating a sphere of radius 0 (see line 636 of proximity_engine.cc from this link https://github.com/RobotLocomotion/drake/blob/master/geometry/proximity_engine.cc#L636)

SeanCurtis-TRI, 2018-09-14:
There's nothing for it but to parse a file. 

My recommendation is as follows:

1. For this pass, only accept .obj files.
2. You'll have to parse the obj file. There's some code that does basically what you want to do; you'll find it [here](https://github.com/RobotLocomotion/drake/blob/master/attic/multibody/shapes/geometry.cc#L237).

If you have any questions, we can discuss it on Friday.

SeanCurtis-TRI, 2018-09-14:
Arguably, we might consider rolling that functionality into FCL. It's a good, fundamental piece of code that would benefit FCL and would prevent us from cluttering up Drake. But, on the other hand, if we want to do TreeViewer, we'd have to parse in Drake and dispatch the vertices....so, for now, let's just do it in Drake.

sherm1, 2018-09-14:
@DamrongGuoy, you could instead test the FCL side next independently by generating some simple Convex meshes by hand and adding unit tests there to ensure that those meshes will work once they get there. Once those work you could return to obtaining meshes from files. Both things have to get done, but I'm thinking it might make testing the ".obj" parsing easier if you know that the meshes should behave properly once they get to FCL. I think either ordering would work though.

@SeanCurtis-TRI, since Drake always has FCL built in I suppose we could add the ".obj" parser to FCL and still use it from within Drake?

SeanCurtis-TRI, 2018-09-14:
On the FCL side there are already a number of tests for the convex shape (written by yours truly). You can find it in [`test_convex.cpp`](https://github.com/flexible-collision-library/fcl/blob/master/test/geometry/shape/test_convex.cpp). So, I believe, you can have pretty reasonable confidence that the convex shape will produce reasonable results.  :)

I realized I pointed you to the wrong section in Drake. Drake already has a dependency on [tiny_obj_loader](https://github.com/syoyo/tinyobjloader) and it is exercised [here](https://github.com/RobotLocomotion/drake/blob/master/attic/multibody/shapes/geometry.cc#L314). Unsurprisingly, it's in the `Mesh::LoadObjFile()` method. D'oh!

DamrongGuoy, 2018-09-17:
Thanks to @sherm1 and @SeanCurtis-TRI.  I checked how `tiny_obj_loader` was used in:
 ```
DrakeShapes::Mesh::LoadObjFile()
(drake/multibody/shapes/geometry.cc).
```
I will use `tiny_obj_loader` to read Obj file and convert to fcl::Convex in:
```
geometry::internal::ProximityEngine::Impl::ImplementGeometry(Convex&) 
(drake/geometry/proximity_engine.cc)
```
The former `LoadObjFile()` does triangulate non-triangular faces, but the latter `ImplementGeometry(Convex&)` will keep polygonal faces for `fcl::Convex(int* faces)`.

PR #9471 (2018-09-27): geometry: Support Convex mesh.
Fix #9443 
1. Add Convex to shape_specification.
2. Skeleton code of ImplementGeometry(Convex) in ReifierTest and ShapeToLcm.
3. ImplementGeometry(Convex) in ProximityEngine by reading Obj file.

WIP.  I still need to set up unit tests and to complete the ImplementGeometry(Convex) in ReifierTest and ShapeToLcm.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9471)
closed 2018-09-27 by DamrongGuoy