RobotLocomotion/drake #9469: Automatically label PRs whose CI results are too stale

jwnimmer-tri, 2018-09-18:
To prevent problems like #9468 (and a few other recent instances like it), we would like to prevent merging PRs when their CI results are too stale -- where `master` has evolved out from underneath them.

We'd like to automatically label PRs whose CI results are too stale -- then we can have Reviewable prevent their merging.  (Or maybe the status / label could be just like any other status-check.)

As a start, CI results older than a couple days could be highlighted.  You could imagine more imaginative checks, that looks at how much has been merged, or if related packages were touched (even a few minutes ago).

(Discussed in `#platform_review` slack.)

Over to @EricCousineau-TRI for prioritization.
\CC @jamiesnape

jamiesnape, 2019-02-05:
Also relates #10582.

jamiesnape, 2019-02-05:
I would probably promote this to medium (and you can assign it to me).

jwnimmer-tri, 2020-12-10:
I wonder if https://github.com/apps/stale would be able to implement this?  It's been on my wishlist to push forward setting up to automatically _close_ out stale pull requests (those without activity for >6 months).  Maybe two birds with one stone.

jwnimmer-tri, 2021-04-16:
https://github.com/RobotLocomotion/drake-external-examples/pull/201 is a tryout in `drake-external-examples`.

jwnimmer-tri, 2021-11-10:
We have not had very many stale-merge-race failures recently.  We can re-open this if we start to see them again.

closed 2021-11-10 by jwnimmer-tri