RobotLocomotion/drake #9475: Fix inverse_kinematics_test for asan builds

BetsyMcPhail, 2018-09-19:
After #9431 was merged, this test fails for asan builds  (e.g. https://drake-cdash.csail.mit.edu/testDetails.php?test=21504406&build=1052590)

BetsyMcPhail, 2018-09-19:
Added 'no_ubsan' tag in #9476

jwnimmer-tri, 2020-04-21:
The CDash link is dead, but I've confirmed it still happens on Ubuntu 18.04:
```console
jwnimmer@cons:~/jwnimmer-tri/drake$ CC=clang CXX=clang++ bazel test --compilation_mode=dbg --config=ubsan //multibody/inverse_kinematics:inverse_kinematics_test --test_tag_filters=
INFO: Analyzed target //multibody/inverse_kinematics:inverse_kinematics_test (0 packages loaded, 0 targets configured).
INFO: Found 1 test target...
FAIL: //multibody/inverse_kinematics:inverse_kinematics_test (see /home/jwnimmer/.cache/bazel/_bazel_jwnimmer/10ee8aea36b37b216ef85df1451b64ad/execroot/drake/bazel-out/k8-dbg/testlogs/multibody/inverse_kinematics/inverse_kinematics_test/test.log)
INFO: From Testing //multibody/inverse_kinematics:inverse_kinematics_test:
==================== Test output for //multibody/inverse_kinematics:inverse_kinematics_test:
Using drake_cc_googletest_main.cc

[==========] Running 7 tests from 3 test suites.
[----------] Global test environment set-up.
[----------] 1 test from InverseKinematicsTest
[ RUN      ] InverseKinematicsTest.ConstructorWithJointLimits
solvers/ipopt_solver.cc:323:20: runtime error: null pointer passed as argument 2, which is declared to never be null
/usr/include/string.h:43:28: note: nonnull attribute specified here
================================================================================
```

PR #13111 (2020-04-22): solvers: Fix memcpy undefined behavior in unconstrained optimization
Calling memcpy on nullptr is undefined.

Closes #9475.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13111)
closed 2020-04-22 by jwnimmer-tri