RobotLocomotion/drake #9496: Update existing System class doxygen to provide block diagram schematics

RussTedrake, 2018-09-23:
using the capability provided by #9490 and documented (once that PR lands) at http://drake.mit.edu/doxygen_instructions.html?highlight=doxygen .

cc @SeanCurtis-TRI

RussTedrake, 2018-09-24:
Note: consider using an image as the system label in some of the primitive examples, to show what's possible/acceptable.

rpoyner-tri, 2021-11-19:
I spent a bit of time fiddling with possible ways of embedding images. Unless I'm missing something, embedding an HTML <img> tag was the intention. That will work, but is not very readable in source code, and raises the new issue of how to host the image data, and whether it needs additional handling at doc-build time. I'm punting for now.

rpoyner-tri, 2021-12-06:
For Diagrams that do not have explicit names set in Export{In,Out}put(), the generated names are unusable for Get{In,Out}putPort(str), since a pointer value in hex gets encoded into the name. Here are some possibilities for documenting them in schematics:

* don't bother creating a schematic (status quo for many Diagrams)
* create a schematic, and indicate by color/typography that the "port names" are actually method calls
* create a schematic, and also introduce string names and the Export*() calls (possibly a breaking change??)

jwnimmer-tri, 2021-12-06:
> ... introduce string names and the Export*() calls (possibly a breaking change??)

I don't think that would be considered as a breaking change (at least, when the sub-systems did not have names assigned).  Can you link to a specific example or two, so we can consider the concrete examples?

rpoyner-tri, 2021-12-06:
Immediate example is `PidControlledSystem` -- no schematic, and no names in current `master`.

jwnimmer-tri, 2021-12-06:
In that case, if the `plant` was previously given a `set_name`, then the wrapped names would have been predictable (not hex), so the change would technically breaking.  Still probably worth making that change, though.  For that system, I think that the diagram's input and output port names should be identical to the wrapped plant's port names (with no extra prefix or suffix) -- except for the "desired state" input port, which we'll need to choose a more careful name for.

rpoyner-tri, 2021-12-08:
Copying here for posterity some of the rules/goals agreed on during early PRs here.

 * list port names that can be used as arguments to Get{In,Out}putPort(str)
 * indicate with color ports that may not always be present
 * move other port annotations to the surrounding text
 * for classes that inherit ports:
   * make sure the base classes have diagrams
   * paste/rename diagrams into the derived classes
 * be consistent in the above guidelines

rpoyner-tri, 2021-12-22:
Another note toward some kind of schematic style guide: color, type style.

Many diagrams use gray to indicate ports that may not exist in some circumstance.
Mbp uses green to indicate ports used with SceneGraph.
Mbp uses gray+italic to indicate variable substitution to produce a group of names.
Manipulation station uses orange+bold to indicate ports that only exist in simulation, but not on a real robot.

Some diagrams use gray+italic on ellipses -- not sure if that is even visible.

What shall we use for optional ports, which always exist, but may be left unconnected? I had a thought of using turquoise, but that choice is arbitrary. I'll have to try it to see how it looks rendered.

sherm1, 2021-12-22:
For ports always present but that don't have to be connected, I'd be inclined just to render them black like other always-extant ports, but mention in the surrounding text that you aren't obligated to connect them. If you do use a different color I think you'll still have to say in the text what that color means -- I doubt many people will remember.

rpoyner-tri, 2021-12-22:
Given that we abandoned the idea of colorizing optional ports, I punted the proposed review pass (which was my main motivation for that). We will likely be back through these improving port names if various #12214 fixes land anyway.

Declaring victory. Thanks all, and happy holidays!

closed 2021-12-22 by rpoyner-tri

RussTedrake, 2021-12-22:
Woohoo!  Thank you!