RobotLocomotion/drake #9497: Cost::Eval and Constraint::Eval should have support for typed vectors

RussTedrake, 2018-09-23:
Currently implementations of costs/constraints which must implement `EvaluatorBase::Eval()` take their arguments only in the form of double vectors.  This means that writing complex constraints requires some ugly contract on the order of the variables, and a lot of boilerplate code to pack/unpack the data.

[DirectCollocationConstraint](https://github.com/RobotLocomotion/drake/blob/master/systems/trajectory_optimization/direct_collocation.cc#L78) is one example (but there are many).

It seems like the BasicVector workflow from the systems framework could likely be used again here.  Discussed briefly with @jwnimmer-tri , and we agreed to post it here to host the discussion.

cc @weigao95 , @hongkai-dai , @EricCousineau-TRI

hongkai-dai, 2018-09-24:
My two cents

1. I tried to factor out the pack/unpack to generic functions in each constraint. One example is in https://github.com/hongkai-dai/drake/blob/rigid_body_tree_multiple_shooting/systems/trajectory_optimization/generalized_constraint_force_evaluator.h#L44~L77. This is probably not the best solution (each constraint still needs to have a generic function to do the packing/unpacking. But the packing/unpacking function can be called repeatedly, such as in https://github.com/hongkai-dai/drake/blob/rigid_body_tree_multiple_shooting/systems/trajectory_optimization/rigid_body_tree_multiple_shooting.cc#L392)
2. I think `BasicVector` is a more elegant solution than the generic function I posed in point 1.. On the other hand, this packing/unpacking is only a problem in the nonlinear cost/constraint. In the linear/second-order-cone/PSD constraint, I do not see the packing/unpacking problem. It might be unnecessary for linear constraints/costs to use BasicVector as well?

weigao95, 2018-09-24:
I am not very familiar with `BasicVector`. I am assuming the workflow is the solver interface takes input as `BasicVector`, while the evaluator takes input from inherited classes like `FrameVector` or `PoseVector` customized to the evaluator (correct me if this is wrong).

I agree with @hongkai-dai that pack/unpack only occurs in non-linear costs/constraints. Maybe we can use a separate evaluator base (something like `NonlinearEvaluatorBase`), and a new `prog.AddNonlinearConstraint` method to deal with constraints with pack/unpack problems. In this way, we don't need to remove any existing code, and we can still use symbolic parsing to construct linear constraints. But it might cause some confusion as the functionality of current `EvaluatorBase` and the new `NonlinearEvaluatorBase` is somehow overlapping.

hongkai-dai, 2018-09-24:
If we are going to create subclass like `FrameVector` or `PoseVector` derived from `BasicVector`, then it feels to me that we need to create a subclass of `BasicVector` for each nonlinear constraint. If this is how we shall use `BasicVector`, then I feel the derived class of `BasicVector` is only "local" to that constraint, so I don't see why we have to create a derived class. On the other hand, the approach suggested in https://github.com/RobotLocomotion/drake/issues/9497#issuecomment-424046329 is simpler without creating a derived class. Here is one example (with pseudo-ish code)
```C++
// This constraint evaluates on x, y, z.
class FooConstraint : public solvers::Constraint {
 public:
  template<typename DerivedX, typename DerivedY, typename DerivedZ>
  VectorX<DerivedX::Scalar> PackInputs(const Eigen::MatrixBase<DerivedX>& x, const Eigen::MatrixBase<DerivedY>& y, const Eigen::MatrixBase<DerivedZ>& z) const {
    VectorX<DerivedX::Scalar> input(num_vars());
    input << x, y, z;
    return input;
  }
  template <typename Derived>
  VectorX<Derived::Scalar> get_x_from_input(const Eigen::MatrixBase<Derived>& input) const {
    return input.head(num_x_);
  }
  private:
    void DoEval(const Eigen::Ref<const Eigen::VectorXd>& input, Eigen::VectorXd* y) const {
      x = get_x_from_input(input);
      y = get_y_from_input(input);
      z = get_z_from_input(input);
      // Do the actual computation on x, y, z.
    } 
}
```
Then we we add the constraint, we just call
```C++
auto constraint = std::make_shared<FooConstraint>();
prog.AddConstraint(constraint, constraint.PackInputs(x_var, y_var, z_var));
```
If we use both `BasicVector` and `Constraint`, then we have two classes to be inherited, namely a `DerivedBasicVector` inherited from `BasicVector`, and a `DerivedConstraint` inherited from `Constraint`, and we need to make sure that `DerivedBasicVector` is only used for `DerivedConstraint`. Using the approach with `PackInputs` method, we only need to keep `Constraint` to be inherited. Besides, we do not need to change the API on `Constraint` or `Cost`, but just to add `PackInputs()` method to the constraint/cost class that need them.

RussTedrake, 2018-09-24:
Seems like something more like this would be better than packing/unpacking with a method?  After all, that's basically what BasicVector is meant for?
```
class MyConstraint : public solvers::Constraint {
protected:
  template <class T>
  void DoTypedEval(const MyBasicVectorForX<T>& x,
                      MyBasicVectorForY<T>* y) const {
    y.theta = 2.*x.phi;
    ...
  };

   void DoEval(const Eigen::Ref<const AutoDiffVecXd>& x,
                      AutoDiffVecXd* y) const {
      DoTypedEval(xvec.SetFromVector(x), yvec);
      y = yvec.CopyToVector();
  };

  void DoEval(const Eigen::Ref<const Eigen::VectorXd>& x,
                      Eigen::VectorXd* y) const {
      DoTypedEval(xvec.SetFromVector(x), yvec);
      y = yvec.CopyToVector();
  };

  XVectorType<double> xvec;
  XVectorType<AutoDiff> ...
}
```

To avoid the copies, i think we could make a relatively small change to `EvaluatorBase` to provide an `Allocate` method (with a default implementation).. so that the solver backends could pass in the typed structures directly?

jwnimmer-tri, 2018-09-25:
I haven't been 100% following this thread enough to understand it, but I also wanted to mention a possibly-related idea.

For named vectors (i.e., classes described in `*.named_vector` source files and put through codegen to make C++ classes), right now they always own their storage, by subclassing BasicVector.  We could also just as easily generate a companion named _view_ class -- where the fields are a reference to someone else's storage, like this:
```c++
template <typename T>
class PendulumStateView {
 public:
  explicit PendulumStateView(Eigen::VectorBlock<VectorX<T>> storage)
    : theta(storage(PendulumStateIndices::kTheta)),
      thetadot(storage(PendulumStateIndices::kThetadot)) {}

  T& theta;
  T& thetadot;
}
```

This way, any double array can be accessed via names, no matter who allocated it.  There are times where this is convenient; we just haven't had the pressing need to write the code for it yet.

If it becomes difficult to control the runtime type of the storage for these variables, the code above could also be written to use a view, without changing the allocation-related code.

RussTedrake, 2018-09-25:
Or, if the named_vector approach is too onerous, you could make a similar "view" class inside a particular constraint implementation (similar to Hongkai's "Pack" method, but giving a more intuitive, imho, use in the constraint method.

It might be worth thinking through the DirectCollocationConstraint in more specific details to see what feels most workable?

hongkai-dai, 2018-09-25:
I am convinced. Both the named_vector or the view_vector approach are easier to use than the "Pack" method, and with the generic `DoTypedEval` function for `MyConstraint`, we do not need to change the API of `Constraint/Cost`.

weigao95, 2018-09-25:
I am concerned with if we need to implement `MyBasicVectorForX<T>::SetFromVector` manually, then it feels to me that we are converting the pack/unpack in original binding creation 
```c++
auto constraint = MyConstraint();
prog.AddConstraint(constraint, a_very_complex_var_list); // Do "packing" here
```
into the pack/unpack in `MyBasicVectorForX<T>::SetFromVector` or the construction of `DirectCollocationStateView`. Using DirectCollocationConstraint as an example, the `SetFromVector` method looks like

```c++
template<typename T>
class DirectCollocationVector {
public:
  void SetFromVector(VectorX<T> input) {
    t = input.segment(0, 1);
    x0 = input.segment(1, num_state_);
    x1 = input.segment(1 + num_state_, num_state_);
    u0 = input.segment(1 + 2 * num_state_, num_input_);
    u1 = input.segment(1 + 2 * num_state_ + num_input_, num_input_);
  }
private:
  int num_state_, num_input_;
  VectorX<T> t, x0, x1, u0, u1;
}
```
and the construction of `DirectCollocationStateView` looks like
```c++
template<typename T>
class DirectCollocationStateView {
public:
  DirectCollocationStateView(VectorX<T> input, int num_state, int num_input) 
  : num_state_(num_state), num_input_(num_input),
    t(input.segment(0, 1)),
    x0(input.segment(1, num_state_)),
    x1(input.segment(1 + num_state_, num_state_)),
    u0(input.segment(1 + 2 * num_state_, num_input_)),
    u1(input.segment(1 + 2 * num_state_ + num_input_, num_input_)) {}
private:
  int num_state_, num_input_;
  VectorX<T> t, x0, x1, u0, u1;
}
```

which is similiar to the original code.
However, if the `DirectCollocationVector<T>::SetFromVector` method or the constructor of `DirectCollocationStateView` can be generated automatically, the burden of pack/unpack is not on programmers. Thus, the solution is to use codegen to automatically produce these "helpers" for constraints?

hongkai-dai, 2018-09-25:
Post our discussion from slack

=============================
hongkai.dai [10:13 AM]
Hi Wei, for your github question, you can created `*.named_vector` source file, and it will generate the headers for you automatically. An example is in `examples/acrobot/acrobot_state.named_vector`. Then after building drake through bazel, you should find a file `acrobot_state.h` in the bazel folder.

Wei Gao [10:25 AM]
I roughly understand the process of codegen. I just want to learn the decision choice. After all, using codegen means programming in another language (*.named_vector in this case) and writing build scripts. It seems a trade-off between different kind of complexities.

===========================
I think using named_vector approach, the effort is to write a short `*.named_vector` file, and then write the build script. The alternative is to create the class `DerivedBasicVector` in the `.cc` file. Personally I think  writing the `*.named_vector` file is easier.

EricCousineau-TRI, 2018-09-25:
For API stability / simplicity, my preference would be to keep `EvaluatorBase` as simple as possible, and assume that the `Binding<>` is handling the proper semantics of the value.

For the internal evaluation code, I think I agree with @jwnimmer-tri's comment, but would in general prefer that we start off by teaching `BasicVector` to act as a view at run-time, rather than a separate type; e.g.
```c++
class BasicVector : public VectorBase {
 public:
  BasicVector(int size)
    : owned_value_(size), value_(owned_value_) {}
  ...

  // Make it a static method, **NOT** an overload.
  static BasicVector AsView(Ref<VectorX<T>> view) {
    return BasicVector(view, std::true_type{});
  }

 private:
  BasicVector(Ref<VectorX<T>> view, std::true_type)
    : value_(view) {}

  VectorX<T> owned_value_;  // This is only used for owned values.
  Eigen::Ref<VectorX<T>> value_;  // All code should only reference this.
};
```
Example views for other cases:
```
auto vars = DirectCollocationStateVector<Variable>::AsView(raw_vars);
auto values = DirectCollocationStateVector<double>::AsView(raw_values);
auto ads = DirectCollocationStateVector<AutoDiffXd>::AsView(raw_ads);

vars.ToVector()
values.ToVector()
ads.ToVector()
```

RussTedrake, 2018-09-25:
If teaching BasicVector to be a view also doesn’t not break/complicate the current BasicVector workfow, then it sounds great.  There are lots of other places in the code that could benefit, too

jwnimmer-tri, 2021-11-11:
Related to #15184.

jwnimmer-tri, 2024-03-16:
Closing as "not planned" -- our current architecture direction is to eschew subclassing of BasicVector, so the downcasting stuff will just evaporate someday anyway.

The `pydrake.common.containers.namedview` is a newer, nicer tool that can help along these lines (when in Python).

closed 2024-03-16 by jwnimmer-tri