RobotLocomotion/drake #9512: Check tolerance of Convex in ProximityEngine

DamrongGuoy, 2018-09-25:
Follow up to #9471, when I set up a unit test for ProximityEngine using Convex, I found that I need to use tolerance 1e-3.  I'd like to check why we cannot use smaller tolerance.

DamrongGuoy, 2020-09-30:
Not needed anymore. Close.

SeanCurtis-TRI, 2020-10-01:
@DamrongGuoy Can you indicate why it's not needed anymore? The answer isn't clear to me.

DamrongGuoy, 2020-10-01:
The reason is no users of drake complained about it. If you'd like it investigated, I'll be happy to reopen.

sherm1, 2020-10-01:
It's not clear to me that users would notice if this caused problems. For example, their simulations might just run slowly because of instability caused by noise in the contact results. But they wouldn't know that this was the cause. I still think we should understand (and ideally fix) the low accuracy here. But I agree the priority could be low since we don't have _obvious_ consequences for this yet (maybe not many people are using convex meshes yet?).

DamrongGuoy, 2020-10-01:
Thank you. I reopen now.

SeanCurtis-TRI, 2020-10-01:
As I've pondered this, it seems that there are three sub-issues at play here:

1. Publishing the fact that tolerance sucks such that customers can make informed decisions. This is part of issue #10907 -- part of my Q4 OKRs to resolve.
2. Investigating *why* the tolerance is bad as it appears to be. That seems to be what the language of this issue is about.
3. Improving precision. Some unknown blob of work (most likely related GJK/EPA).

As @DamrongGuoy indicated, no one has complained about apparent precision woes. But as @sherm1 has pointed out, they may not know to complain.

It seems reasonable to close this issue in favor of #10907. With information, customers might be more able to complain (although we don't have many users that *use* `Convex`).

I would advocate that if we keep this issue open, we need to elaborate the issue summary. A link to the PR doesn't seem sufficient -- it would require a great deal of spelunking by future readers to understand what the actual code in question is. Leaving the link as *support* of the issue description is fine, but the actual issue should be spelled out more explicitly.

DamrongGuoy, 2020-10-06:
I agree with Sean.  Close this issue in favor of #10907.

closed 2020-10-06 by DamrongGuoy