RobotLocomotion/drake #9518: Problem Building From Source

dkout, 2018-09-26:
Hi, 

I am unfortunately not able to build Drake from source on my Ubuntu machine.

## Reproduction

After running 'sudo ./setup/ubuntu/16.04/install_prereqs.sh' , when typing: ` bazel build //...`, the project fails to build.

> If your changes to `drake` are more comprehensive, please try to isolate your
problem to the above amount, ensuring that you are using a branch as close to
`master` as possible.

## Debug Information
Using freshly cloned Master branch.

* What operating system are you running Drake on?
(e.g. Ubuntu 16.04, macOS 10.13)

    I am running on Ubuntu 16.04 Xenial.

* Is this an issue with (a) C++, (b) Python, or (c) the distributed tools
(e.g. `drake-visualizer`)?

    Not sure.

* Are you (a) building `drake` from source, or (b) from a binary release?

    From source

* What version of `python` are you using?
+ which python2
/usr/bin/python2
+ python2 --version
Python 2.7.12

* If you are building from source or using the `drake` C++ library as an
external:

    * What version of Bazel and CMake are you using?

        >
+ which bazel
/usr/bin/bazel
+ bazel version
Build label: 0.16.1
Build target: bazel-out/k8-opt/bin/src/main/java/com/google/devtools/build/lib/bazel/BazelServer_deploy.jar
Build time: Mon Aug 13 13:43:36 2018 (1534167816)
Build timestamp: 1534167816
Build timestamp as int: 1534167816
+ which cmake
/usr/bin/cmake
+ cmake --version
cmake version 3.5.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).

    * If you are using Bazel, please provide the host settings in your present
    project workspace:

        INFO: Analysed target @drake//common:print_host_settings (1 packages loaded).
INFO: Found 1 target...
Target @drake//tools/cc_toolchain:print_host_settings up-to-date:
  bazel-bin/external/drake/tools/cc_toolchain/print_host_settings
INFO: Elapsed time: 1.029s, Critical Path: 0.06s
INFO: 1 process: 1 linux-sandbox.
INFO: Build completed successfully, 5 total actions
INFO: Running command line: bazel-bin/external/drake/tools/cc_toolchain/print_host_settings external/drake/tools/cc_toolchain/capture_cINFO: Build completed successfully, 5 total actions
+ capture_cc_env=external/drake/tools/cc_toolchain/capture_cc.env
+ source external/drake/tools/cc_toolchain/capture_cc.env
++ BAZEL_CC=/home/dkout/anaconda3/bin/gcc
+ /home/dkout/anaconda3/bin/gcc --version
gcc (crosstool-NG fa8859cb) 7.2.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

    * If you are using CMake, please provide the compilers being used in your
    project:

        > First, in your build directory, execute
        `cmake -LA <path_to_source_dir> | grep 'CMAKE_.*_COMPILER'`
WARNING: Running Bazel server needs to be killed, because the startup options are different.
WARNING: --batch mode is deprecated. Please instead explicitly shut down your Bazel server using the command "bazel shutdown".
fatal: Not a git repository: '/home/dkout/Interlock/drake3/drake/.git'
CMAKE_CXX_COMPILER:FILEPATH=/home/dkout/anaconda3/bin/c++
CMAKE_C_COMPILER:FILEPATH=/home/dkout/anaconda3/bin/cc

        > Given these values, execute `<compiler> --version`, and provide its
        output.
cc (crosstool-NG fa8859cb) 7.2.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

* If you are building `drake`
[from source](https://drake.mit.edu/from_source.html):

    * What git commit of `drake` are you using?

        > c0b8e42

    * Are you building `drake` with Bazel or CMake?

        >Bazel

    * Are you using `drake` directly, or consuming it as an external?

        >directly

    * If an external, is it most like
    [`drake_bazel_external`](https://github.com/RobotLocomotion/drake-shambhala/tree/master/drake_bazel_external) or
    [`drake_cmake_external`](https://github.com/RobotLocomotion/drake-shambhala/tree/master/drake_cmake_external)?

        > Replace this with your answer.

* If you are using `drake` [from a binary release](https://drake.mit.edu/from_binary.html):

    * What are the contents of `.../drake/share/doc/drake/VERSION.txt`?

        > Replace this with your answer.

    * What URL did you download `drake` from?

        >git@github.com:RobotLocomotion/drake.git

    * Are you writing a CMake project which is using `drake`, like
    [`drake_cmake_installed`](https://github.com/RobotLocomotion/drake-shambhala/tree/master/drake_cmake_installed)?

        > No

dkout, 2018-09-26:
That was my bad, forgot that exporting to path doesn't save. However I am still getting this  following error:
`bazel build //... --python_path=/usr/bin/python2
Starting local Bazel server and connecting to it...
INFO: Analysed 6816 targets (337 packages loaded).
INFO: Found 6816 targets...
ERROR: /home/dkout/Interlock/drake3/drake/drake/attic/multibody/parsers/BUILD.bazel:25:1: C++ compilation of rule '//attic/multibody/parsers:parsing' failed (Exit 1)
In file included from bazel-out/k8-opt/bin/attic/multibody/parsers/_virtual_includes/parsing/drake/multibody/parsers/sdf_parser.h:9:0,
                 from attic/multibody/parsers/sdf_parser.cc:1:
bazel-out/k8-opt/bin/attic/multibody/parsers/_virtual_includes/parsing/drake/multibody/parsers/parser_common.h:6:10: fatal error: tinyxml2.h: No such file or directory
 #include <tinyxml2.h>
          ^~~~~~~~~~~~
compilation terminated.
INFO: Elapsed time: 81.335s, Critical Path: 15.65s
INFO: 66 processes: 66 linux-sandbox.
FAILED: Build did NOT complete successfully
`

soonho-tri, 2018-09-26:
Can you check if you have `/usr/include/tinyxml2.h`? Also please check the output of `dpkg -l libtinyxml2-dev`.

jwnimmer-tri, 2018-09-27:
Hm.  The file generated by `cps2cmake` happens via shell redirection, instead of `open(something, 'w')`.  As such, it is vulnerable to any `import`'ed module printing a diagnostic and corrupting the output.  Since the exitcode would still be zero, the file would not be recreated, and the error would persist.  I wonder if that's what happened here.

jamiesnape, 2018-09-28:
> Hm. The file generated by cps2cmake happens via shell redirection, instead of open(something, 'w'). As such, it is vulnerable to any import'ed module printing a diagnostic and corrupting the output. [...]

I think that could be true, but I would only expect that to affect the generated `-config.cmake` file that is not used by Bazel.

soonho-tri, 2018-09-28:
@dkout has [anaconda](https://www.anaconda.com) installed in his system. From his log, we can tell his `PATH` included `/home/dkout/anaconda3/bin` and Anaconda's python3 is selected when `python` is run (while `/usr/bin/python2` is picked up). I was wondering if this caused the problem or not. If so, I guess we need to update the issue template to provide the outcome of `which python` in addition to `python2`. @jamiesnape , do you think it could be one of the reason why he had a problem?

jamiesnape, 2018-09-28:
Certainly most of the issues that we get from users involve an `anaconda` install. We can probably find a way to make drake error out if `python` = `python3`. I don't think we allowed for that eventuality   yet.

soonho-tri, 2018-09-29:
@dkout , I'm closing this one assuming that you don't have further issues in building Drake. If not, please re-open it.

> Certainly most of the issues that we get from users involve an anaconda install. We can probably find a way to make drake error out if python = python3. I don't think we allowed for that eventuality yet.

@jamiesnape , thanks for the explanation. Please feel free to open an issue if you want.

closed 2018-09-29 by soonho-tri