RobotLocomotion/drake #9523: Odd namespace requirement for SystemTypeTag

sherm1, 2018-09-26:
Working on #9489 I discovered the hard way that `systems::SystemTypeTag<SomeTemplateType>`, where `SomeTemplateType` is not in the `drake::systems` namespace, fails drastically in CI. Strangely it works fine when I build locally (in CLion). The message is roughly "SomeType is not a template class but the parameter is a template template parameter" (sorry I don't have the original now).

The fix turned out to require adding the missing namespaces to the template argument. So while 
`systems::SystemTypeTag<MultibodyPlant>` failed, `systems::SystemTypeTag<multibody::multibody_plant::MultibodyPlant> succeeded even though the statement itself was already in `namespace multibody_plant`.

It would be nice if these superfluous namespaces were not required, but at least the (otherwise excellent) [documentation](http://drake.mit.edu/doxygen_cxx/group__system__scalar__conversion.html) should *emphasize* that they are. (I see now that the examples there actually do show the extra namespace but I didn't catch on that those were required, misled I suppose by the fact that my local builds worked fine!)

jwnimmer-tri, 2018-09-26:
I am surprised it ever worked for you.  It has always yelled the first time, for me.  I guess, though, when I last worked on this, we were still using clang as our default compiler.  It sounds like GCC (our default in present day) is not complaining here, even though the program is ill-formed.

I wonder if gcc-7 complains?  If so, I would be inclined to WONTFIX this.

jwnimmer-tri, 2018-09-26:
Nope; gcc-7 is still happy when the namespace qualifier is missing.

(BTW You don't need `multibody::multibody_plant::`, just `multibody_plant::` would be enough.)

jwnimmer-tri, 2018-09-26:
So actually, clang-4 (what we have in CI for now) yells when the namespace qualifier is missing, but clang-6 does not yell.  So perhaps the program is well-formed, but clang-4 is defective.  Since this is a clang-4 specific issue, and we're dropping support for it (#9483) momentarily, I'll say WONTFIX to a doc update.

sherm1, 2018-09-26:
The discussion [here](https://stackoverflow.com/questions/17687459/clang-not-accepting-use-of-template-template-parameter-when-using-crtp) suggests to me that clang is broken, rather than gcc. Why do you think the program is ill-formed without the namespace qualifiers?

jwnimmer-tri, 2018-09-26:
You are correct.

sherm1, 2018-09-26:
Our comments overlapped. Rather than WONTFIX, maybe we need to remove the spurious `sample::` namespaces from the documentation? But currently things won't work if we don't show that. I think either way the documentation needs ... something.

jwnimmer-tri, 2018-09-26:
Once we're on clang-6, I have no objection to that.

jwnimmer-tri, 2018-09-26:
Just to forestall any misunderstanding... I think the action here is to wait until #9483 merges, and then to remove the, e.g., `sample::` namespaces inside the `SystemTypeTag<...>` uses in the scalar conversion doxygen, and also remove the analogous uses throughout the code (such as `multibody_plant::`) so that the codebase is consistent with the documentation.

sherm1, 2018-09-27:
Is that imminent and does that mean no one will have this problem after that? Are Macs still using broken clang versions? Unless this is never going to bite anyone again, I would prefer to add a note now to the documentation saying that the namespaces are mandatory when compiling with clang versions < 6.0.

jwnimmer-tri, 2018-09-27:
Once that PR merges, my understanding is that all of our supported compilers will be safe with respect to this false positive.  Its hard to predict a timeline, but I would think I'd merge within week or so.

jwnimmer-tri, 2018-10-10:
Yup.  The required PR merged 9 days ago, so indeed this is just a "remove some stray namespace ornaments" fixup now.

PR #12284 (2019-10-30): Remove unncessary namespace qualifier for SystemTypeTag values
Closes #9523.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12284)
closed 2019-10-30 by jwnimmer-tri