RobotLocomotion/drake #9553: geometrty: Improve data representation of convex meshes in ProximityEngine

DamrongGuoy, 2018-09-28:
We would like to improve data representation of convex meshes in ProximityEngine. The data is needed by fcl::Convex.  The improvement includes
1. Change `ConvexData` from a `struct` to a `shared_ptr` of `struct`.
2. Change `convex_objects_` from a `vector` of `ConvexData` to a `map` from `filename` to `ConvexData`.
3. Copy `convex_objects_` in the copy constructor of ProximityEngine<T>::Impl.

Improvement 1 is in anticipation of the future fcl::Convex constructor that will take `shared_ptr.`
Improvement 2 will optimize instantiation of multiple objects of the same shape from the same file.
Improvement 3 is to prevent a potential problem introduced by #9471.

jwnimmer-tri, 2024-03-16:
FYI @SeanCurtis-TRI per your recent work on convex & fcl.

jwnimmer-tri, 2024-03-20:
Actually, if we haven't cared about this in the past 6 years, I'll venture a guess that it's not important anymore, so closing as "not planned".

closed 2024-03-20 by jwnimmer-tri

SeanCurtis-TRI, 2024-03-20:
Ironically, it's in the midst of being kind of addressed with #21147.

`Mesh` and `Convex` now have a unique representation of their convex hull which has a direct mapping to `fcl::Convexd`. The only thing we're really missing is a geometry cache so we don't have to parse the same file over and over. But that's a known, long-lasting thing that doesn't need an issue.