RobotLocomotion/drake #9555: Provide symbolic substitution (Expr -> Expr)

soonho-tri, 2018-09-28:
For now, we only allow "Var -> Expr" case.

/cc @shensquared , @hongkai-dai

jwnimmer-tri, 2021-11-09:
Note that #16014 (pydrake only) is related to this.

jwnimmer-tri, 2021-11-18:
In pydrake, we have `Unapply` now (see above).

For C++ users, we have `VisitExpression` which the user can call using their own visitor that _could_ be used to implement structural expression pattern matching and substitution, as one example.  Possibly that basic architecture is sufficient for the need?

@shensquared or @hongkai-dai is there some use case you have in mind that is not yet supported, or is overly difficult?  Please let me know what's still missing, if anything.

hongkai-dai, 2021-11-18:
Thanks Jeremy, I think this looks good. Personally I haven't done symbolic substitution using Expr -> Expr yet, but this feature looks nice (especially in pydrake).

shensquared, 2021-11-19:
Hey, Jeremy! IIRC, this issue was created out of a discussion very similar to Alexandre's request on [Stackoverflow](https://stackoverflow.com/questions/69774737/xreplace-in-drake-symbolic); essentially a change of (external) variable to comply with the SOS `form'.  So your PR and unittest example addressed that; thanks!

jwnimmer-tri, 2021-11-19:
Okay, thanks!  I will close this for now, since it seems most active interest is from the pydrake side.

If a C++ Drake user finds the issue in the future and would like to have `Unapply` sugar built-in to C++ (instead of needing to roll your own visitor), please speak up and we can re-open.

closed 2021-11-19 by jwnimmer-tri