RobotLocomotion/drake #9560: Needed: JacobianDotTimesV methods in MultibodyTree

RussTedrake, 2018-09-30:
@pangtao22 specifically asked about the analog to `rigidbodytree.CalcBodySpatialVelocityJacobianDotTimesVInWorldFrame`

I know understand that this is blocking our ability to release the next pset with MBT/P.

@pangtao22 -- please respond here being very explicit about what the request is?

cc @tlpmit

pangtao22, 2018-10-01:
#9311

pangtao22, 2018-10-01:
We need 

- [ ] A MBT method similar to `rbt.JacobianDotTimesV`

- [ ] Python bindings for `EvalBodySpatialVelocityInWorld`, which I assume is similar to `rbt.RelativeTwist`?

- [ ] Python bindings for `SpacialVelocity` and `SpacialForces`, so that we can call `inverseDynamics` with nonzero external forces.

amcastro-tri, 2018-10-01:
Thanks @pangtao22, I'll see what I can deliver quickly regarding `JdotTimesV`.

amcastro-tri, 2018-10-01:
@pangtao22, regarding passing forces to ID, what kind of forces would you need to be able to apply first?
1. Joint generalized forces (torques for revolute joints for instance).
2. Spatial forces on bodies.

pangtao22, 2018-10-01:
Ideally we would need both. 
- Joint generalized forces (torques for revolute joints for instance) can be used to model the external torque command sent to the iiwa arm through lcm.
- Spatial forces on bodies are used to model external forces acting on iiwa's gripper.

tri-ltyyu, 2018-11-09:
Seems like #9642 is merged, should this be marked complete, @amcastro-tri ?