RobotLocomotion/drake #9564: Add regression test to prevent unplanned removal of public API

EricCousineau-TRI, 2018-10-01:
It'd be nice to have a record of (a subset of) the public symbols that Drake offers, and have a small regression test that will fail if a public symbol is removed / renamed without a plan.

It wouldn't be perfect, in that it would require manual update, as we do with `build_components`, 
For C++, it'd be easy to leverage the stuff in `mkdoc.py` to generate that list.
For Python, it may take a bit more time to figure out the minimal set of symbols (e.g. not requiring that `pydrake.__init__` have `sys` imported).

Example case:
#9489 removed `MultibodyTree::AllocateDefaultContext`, which broke downstream code usage in Anzu. It'd have been nice to put a test that would automatically complain about the loss of this public API member.

jwnimmer-tri, 2018-10-01:
I'm not sure the problem with #9489 was that the author and reviewers were unaware that an API was being removed.  I'm not sure that this acute proposal will add much value.

sherm1, 2018-10-01:
Right. At least for me I didn't realize this was already in use outside of Drake. Even so I would have left it there deprecated but couldn't figure out a way to make it work since an MBTree shouldn't be able to allocate System resources without a System present. @EricCousineau-TRI's workaround still requires a change to existing code -- it keeps the API but changes the circumstances in which it may be called.

I think Drake API wranglers like me should start building Anzu. In that case I would have prepared a PR to upgrade Anzu to the new API.

jamiesnape, 2018-10-01:
> At least for me I didn't realize this was already in use outside of Drake.

Could we ever know? The best we can probably do here is offer a fixed deprecation period for any change, I think.

jamiesnape, 2018-10-01:
Or maybe a slightly less friendly, but useful error message or comment with migration instructions when that is not possible, given the major refactoring that happens any given day here.

EricCousineau-TRI, 2020-07-21:
I'm going to close this for now. We have some brief docs about testing with `drake-external-examples`:
https://drake.mit.edu/downstream_testing.html

jamiesnape, 2020-07-21:
I guess I will fix this during #12783 anyway, but it is anything but solved. `drake-external-examples` tests the mechanics of the build, not the API.

EricCousineau-TRI, 2020-07-21:
Aye, will re-open for now. (And sorry Jeremy - I didn't look at the issue ownership :( )

jamiesnape, 2020-07-21:
FTR https://manpages.debian.org/testing/dpkg-dev/dpkg-gensymbols.1.en.html would diff the API from month-to-month (or however often we build packages).

jwnimmer-tri, 2021-11-11:
I am a bit interested in running symbol linters as part of the release process, but moreso about leaking private symbols accidentally, than trying to carefully guard what public ABI symbols we export.  Code reviews seem to working OK for that purpose for now.

closed 2021-11-11 by jwnimmer-tri