RobotLocomotion/drake #9566: multibody: Support loop joint topology

EricCousineau-TRI, 2018-10-01:
- [ ] Support for loop joint topology in MBT / MBP API
- [ ] Support for loop joint specification in SDF parsing

sherm1, 2020-05-12:
Drake's [bushing element](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_linear_bushing_roll_pitch_yaw.html) is available as a workaround until this is done.

sherm1, 2021-04-18:
For implementation notes, see #1867.

RussTedrake, 2023-06-16:
Bread crumb: The SAP solver now supports closed loop topologies via constraints.  See, for instance [MultibodyPlant::AddBallConstraint()](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#a0572ce9897e3727ccdfd5fbb34e6512a) and/or [MultibodyPlant::AddDistanceConstraint()](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#a8f29f6aa01dd68ee2babe09ace936ebf)

RussTedrake, 2023-06-16:
See also https://github.com/RobotLocomotion/drake/issues/18803  for the parsing.  (I actually think we could probably close this issue given the coverage from the other two).