RobotLocomotion/drake #9569: mosek tests failing on my mac

RussTedrake, 2018-10-02:
Just refreshed my license file, but one test is still failing with:

```
solvers/test/mosek_solver_test.cc:129: Failure
Value of: spruce::path(log_file).exists()
  Actual: true
Expected: false
solvers/test/mosek_solver_test.cc:133: Failure
Value of: spruce::path(log_file).exists()
  Actual: true
Expected: false
Problem
  Name                   :
  Objective sense        : min
  Type                   : LO (linear optimization problem)
  Constraints            : 1
  Cones                  : 0
  Scalar variables       : 2
  Matrix variables       : 0
  Integer variables      : 0

Optimizer started.
Presolve started.
Linear dependency checker started.
Linear dependency checker terminated.
Eliminator started.
Freed constraints in eliminator : 1
Eliminator terminated.
Eliminator - tries                  : 1                 time                   : 0.00
Lin. dep.  - tries                  : 1                 time                   : 0.00
Lin. dep.  - number                 : 0
Presolve terminated. Time: 0.00
Optimizer terminated. Time: 0.00

solvers/test/mosek_solver_test.cc:137: Failure
Value of: spruce::path(log_file).exists()
  Actual: true
Expected: false
[  FAILED  ] MosekTest.TestLogFile (5 ms)
```

hongkai-dai, 2018-10-09:
Russ, does `gurobi_solver_test` work on your Mac machine? It seems it's the problem of `spruce`.

jwnimmer-tri, 2018-10-09:
It is more likely some combination of `temp_directory` not working, or the bazel sandbox being disabled, or something.  The test first checks that there is no log file, prior to asking mosek to make a log file.  Somehow, at the start of the test, there was a log file already.  Something about the testing environment was corrupted with local files.  Since CI does not show this problem, I am not sure there is much to do, other than debug Russ's mac directly, to find why `temp_directory() + "mosek.log"` already exists when the test starts up.

RussTedrake, 2018-10-09:
confirmed that `gurobi_solver_test` passes (and `mosek_solver_test` still fails).

RussTedrake, 2018-12-21:
fwiw -- this happens for me on my ubuntu machine, too.

PR #10297 (2018-12-21): fix path to mosek test log file
Resolves #9569

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10297)
commit 5f353295 (2018-12-21): fix path to mosek test log file

Resolves #9569
closed 2018-12-21 by RussTedrake