RobotLocomotion/drake #9572: multibody parsing file organization

RussTedrake, 2018-10-02:
`multibody/parsers` is an empty folder -- for deprecation.  fine.
`multibody/parsing` has content, but not content that parses into MBP.
`multibody/multibody_tree/parsing` has the parsers I was looking for.

But the namespace in that folder is
```
namespace drake {
namespace multibody {
namespace parsing {
```
which is exactly the folder that I didn't want.  huh? 

can we please clean this up?

EricCousineau-TRI, 2019-03-25:
I think this was resolved by @jwnimmer-tri's cleanup efforts (both `attic/` and `parsing/` consolidate).

@jwnimmer-tri Do you think we can close this?

jwnimmer-tri, 2019-03-25:
I'll double-check and then either fix with another PR or close if it's already done.

closed 2019-03-25 by jwnimmer-tri