RobotLocomotion/drake #9574: mkdoc: No docstring for `RigidBodyFrame::RigidBodyFrame<T>`

EricCousineau-TRI, 2018-10-03:
At a1bfdb4, this is the generated pybind documention I see for `RigidBodyFrame::RigidBodyFrame<T>`:
```c++
    // Symbol: RigidBodyFrame::RigidBodyFrame<T>
    struct /* ctor */ {
      // Source: drake/multibody/rigid_body_frame.h:32
      const char* doc = R"""()""";
      // Source: drake/multibody/rigid_body_frame.h:32
      const char* doc_2 = R"""()""";
      // Source: drake/multibody/rigid_body_frame.h:41
      const char* doc_3 = R"""()""";
      // Source: drake/multibody/rigid_body_frame.h:50
      const char* doc_4 = R"""()""";
      // Source: drake/multibody/rigid_body_frame.h:60
      const char* doc_5 = R"""()""";
    } ctor;
```

However, the source code indicates that there should be docstrings:
https://github.com/RobotLocomotion/drake/blob/a1bfdb4af526c8d5b4574fca3329911239c9c2ac/attic/multibody/rigid_body_frame.h#L32

\cc @jamiesnape

jamiesnape, 2018-10-03:
This is indeed not the only place. Not sure having anything between a Doxygen comment and its documented entity is really correct anyway, though Doxygen usually tolerates it.

EricCousineau-TRI, 2019-01-22:
Closing this; will be better addressed by #9745 as Jamie mentioned.
(Also, going to spend zero effort fixing `attic/` bits unless a user brings it up as a pain point. This was proactive on my point.)

closed 2019-01-22 by EricCousineau-TRI