RobotLocomotion/drake #9598: Bug in CalcPointsGeometricJacobianExpressedInWorld()

amcastro-tri, 2018-10-05:
[CalcPointsGeometricJacobianExpressedInWorld()](https://github.com/RobotLocomotion/drake/blob/master/multibody/multibody_tree/multibody_tree.h#L1736) assumes that the input frame is a body frame. This is a bug.

Action items:
- Fix code to remove this assumption.
- Implement unit tests.
- Update docs to clearly reflect this, as done in #9597 for bias terms.

amcastro-tri, 2018-10-07:
Update, is not a bug, but a defect on the documentation.

amcastro-tri, 2018-11-12:
This is part of the Epic to finish the family of Jacobians.

sherm1, 2020-06-15:
@amcastro-tri @mitiguy can this be closed now?

mitiguy, 2020-06-16:
@sherm1 and @amcastro-tri  I think this issue is now closed.
I think this was issue was addressed in connection with issue #12600
Remove deprecated method CalcFrameGeometricJacobianExpressedInWorld()
and instead use CalcJacobianSpatialVelocity().

From a comment in issue #12600:  It seems GeometricJacobian and AnalyticalJacobian now only occupy a poor spot in the old deprecated RigidBodyTree attic.
This issue follows deprecation of other Jacobian methods from issue #12154 and PR#12171.

closed 2020-06-16 by mitiguy