RobotLocomotion/drake #9600: pydrake doc: gcc and clang disagree on need for lambda capture for `constexpr` vars

EricCousineau-TRI, 2018-10-05:
From here: https://github.com/RobotLocomotion/drake/pull/9571#issuecomment-426708850

Workarounds:
- Use `[&]` in lambda capture
- Add `copts = ["-Wno-unused-lambda-capture"]` for `pybind_py_library`

Action items:
- [ ] Use a workaround in relevant sites (searching for `doc_lambda`)
- [ ] Find or file upstream bug report

\cc @m-chaturvedi @jamiesnape

EricCousineau-TRI, 2018-10-05:
Possibly related; however, I am getting this with `clang-6.0`, and it'd seem that this should have been incorporated already:
https://bugs.llvm.org/show_bug.cgi?id=31815

m-chaturvedi, 2018-10-05:
Maybe related: https://reviews.llvm.org/D33526

EricCousineau-TRI, 2018-10-05:
Seems that this patch was incorporated before `clang-6.0`:
```
$ clang-6.0 --version
clang version 6.0.0-1ubuntu2~16.04.1 (tags/RELEASE_600/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```

The branch in the Git repo `release_60`:
https://github.com/llvm-mirror/clang/tree/release_60
Does incorporate the corresponding git commit for the patch:
https://github.com/llvm-mirror/clang/commit/6c7060619c4c2c8195dfd974136ab9cf9bac42e9

So perhaps it's still an issue?

EricCousineau-TRI, 2018-10-05:
Huh, seems like you can also change `auto& doc = ...` to `constexpr auto& doc = ...`, and then neither `clang` nor `gcc` seem to care.

jamiesnape, 2018-10-05:
Very Java-like (except that would be `final`). Makes a certain amount of sense.

EricCousineau-TRI, 2018-10-05:
Reproduction code + output:
https://github.com/EricCousineau-TRI/repro/blob/c80dc2360068a256665275c52518ed2876f5d112/bug/clang_unused_lambda_capture/output.txt

EricCousineau-TRI, 2018-10-06:
Filed bug with `clang`, will see if they know where it should go: https://bugs.llvm.org/show_bug.cgi?id=39202

EricCousineau-TRI, 2018-10-08:
From chat with @m-chaturvedi, seems that GCC 7.3.0 on Bionic decided to further deviate from `clang`; with the merge of #9606, we now get a Bionic failure:
https://drake-jenkins.csail.mit.edu/job/linux-bionic-gcc-bazel-experimental/11/console
```
bindings/pydrake/systems/framework_py_semantics.cc:72:37: note: the lambda has no capture-default
   auto bind_common_scalar_types = [m](auto dummy) {
                                     ^
bindings/pydrake/systems/framework_py_semantics.cc:31:19: note: 'constexpr const<unnamed struct>::<unnamed struct>::<unnamed struct>& doc' declared here
   constexpr auto& doc = pydrake_doc.drake.systems;
                   ^~~
bindings/pydrake/systems/framework_py_semantics.cc:310:12: error: 'doc' is not captured
            doc.Parameters.SetFrom.doc);
```

Will submit a chaser PR to use the second workaround :(

EricCousineau-TRI, 2018-10-08:
I may have spoken too soon? I am unable to reproduce this with simple reproduction:
https://github.com/EricCousineau-TRI/repro/commit/4109dc4b2796366e085ebd94bf29255e690e1e4b?diff=split#diff-bc917a88a9cbeb890e61e5fb340a8ec5

Even trying to re-simulate the nested structs does not yield an issue:
https://github.com/EricCousineau-TRI/repro/commit/bbc8f78c14b1bc08151f45bad883d86405ecf6a7?diff=split#diff-bc917a88a9cbeb890e61e5fb340a8ec5R18

Not sure what the discrepancy is... @m-chaturvedi @jamiesnape Might I ask if y'all have an inkling?

EricCousineau-TRI, 2018-10-08:
Mmanu found that this error can be reproduced if we use a generic / templated lambda, such as using `auto` as a parameter:
https://github.com/EricCousineau-TRI/repro/commit/fd025bc?diff=split#diff-bc917a88a9cbeb890e61e5fb340a8ec5

Seems related to old aforementioned bug, just an uncovered corner case: https://bugs.llvm.org/show_bug.cgi?id=31815

PR #9606 (2018-10-08): pydrake doc: Use `constexpr auto&` for nesting aliases
Resolves #9600

\cc @m-chaturvedi 

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9606)
PR #9630 (2018-10-09): pydrake: Readdress inconsistent storage duration btw g++ and clang
Closes #9600.
(Again)

\cc @jamiesnape 

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9630)
closed 2018-10-09 by EricCousineau-TRI