RobotLocomotion/drake #9623: multibodyplant needs `get_state_vector(model_instance)`

RussTedrake, 2018-10-09:
and, of course, also the mutable version.

As discussed in the manipulation_station example PR (#9596)

RussTedrake, 2018-10-09:
I *believe* that the current best implementation is what I found in the model_instance version of `CopyContinuousStateOut` method of MultibodyPlant.

RussTedrake, 2018-10-09:
fwiw -- this does not quite work for me:
```
  plant_->tree().set_velocities_in_array(
      iiwa_model_, v, &(plant_->tree()
                            .get_mutable_multibody_state_vector(&plant_context)
                            .head(plant_->num_positions())));
```
because I can't take a reference of the temporary object.

sammy-tri, 2018-10-09:
Taking ownership per slack discussion.  I think the actual new functions will be `MultibodyTree::get_multibody_state_vector()` and mutable with an overload to accept a `ModelInstanceIndex`.

amcastro-tri, 2018-10-09:
If you allow me, could I throw alternative signatures in? I think having `MultibodyTree::get_multibody_positions(model_instance)` and `MultibodyTree::get_multibody_velocities(model_instance)` would also be super helpful (with also mutable setters).

sammy-tri, 2018-10-09:
Does this also imply a version for the entire tree?  And why is the word `multibody` in the function name?  Is MBT going to be holding some other kind of positions/velocities/state vectors?

amcastro-tri, 2018-10-09:
> Does this also imply a version for the entire tree? 

I believe that would be helpful also. In the same way right now we have `MBT::get_multibody_state_vector()`.

> And why is the word multibody in the function name?

sorry, mostly copy/pasta. Something like `MultibodyTree::get_velocities(model_instance)` should suffice.

The distinction is purposely made with `get_multibody_state_vector()` since our state might include other stuff in the future (energy tracking to name an easy one).

sammy-tri, 2018-10-09:
Having done some digging into this, I believe the mutable versions would be feasible only under specific configurations, namely that the world has only a single kinematic chain of joints (I think you can have as many things in the world as you want if they don't have anything further outboard).  In any other configuration (say, two iiwa arms attached to the world) the positions/velocities for each model instance will not be contiguous in the state vector of the tree.  #9455 illustrates one example of this, as does the implementation of `ModelInstance<T>::set_positions_in_array`.

There might be a way to get a scatter/gather eigen thing which has the mutable semantics of `Block`, but I can't seem to find it.  I think the best I can do is something like 
```
void set_model_instance_positions(ModelInstanceIndex index,
                                  const Eigen::Ref<const VectorX<T>>& model_q,
                                  systems::Context<T>* context) const;
```
which is basically sugar around `set_positions_in_array`.

Thoughts?

jwnimmer-tri, 2018-10-09:
I think you're right that Eigen doesn't offer a gather type.  I think the best you could do is offer a getter than returns a `std::unique_ptr<drake::systems::Supervector<T>>` that the user could call `result->SetFromVector(Eigen::Ref<>)` on.  That doesn't seem to offer any benefit though -- I'd say to just stick with the setter(s).

edrumwri, 2018-10-09:
I vote for sticking with the setters.

amcastro-tri, 2018-10-10:
I agree with @edrumwri, setters are simpler and get the job done.

RussTedrake, 2018-10-21:
Ping on this one.  My code is getting more and more complex without it (needing to hard code link names) as I’m putting more objects into the manipulation station demo.

edrumwri, 2018-10-21:
I'm in the midst of adding similar functionality. I can add this at the
same time.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

edrumwri, 2018-10-26:
@RussTedrake I've had this issue in the queue for resolving earlier this week but #9707/#9702 (earlier in the queue) is proving much trickier than anticipated. Please ping if you'd like a priority change.

sherm1, 2019-01-05:
@edrumwri do you want to close this issue and open one just for the setters?

edrumwri, 2019-01-05:
Addressed *in full* by PR #10017.

PR #9977 (2018-11-12): Added multibody state accessors and mutators.
Addresses issue #9623.

EDIT (eric):
Resolves #9623
Resolves #9762

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9977)
closed 2019-01-05 by edrumwri