RobotLocomotion/drake #9624: Gurobi v8.0.0 dependency makes Python version of Drake unusable.

vBarbaros, 2018-10-09:
In the Make files for drake build the dependency for Gurobi is specified as for v8.0.0, while Gurobi now is releasing v8.0.1, and they are not supporting v8.0.0 anymore (just confirmed with them).

Is there a solution?

I tried to modify some entries in the Make files, but not sure if I caught all the places, since the problem still persist - I can't run the build commands:
`cmake ../drake`
`make -j`

Thanks

jwnimmer-tri, 2018-10-09:
@vBarbaros Thanks for the report.  Yes, it looks like the Gurobi website only offers v8.0.1 now.  We should update Drake to use 8.0.1 by default.  @EricCousineau-TRI over to you for prioritization.

@vBarbaros The files that will need to be changed are within the https://github.com/RobotLocomotion/drake/tree/master/tools/workspace/gurobi directory.  You might be able to tweak them yourself to use 8.0.1, in your local build, for now.

jamiesnape, 2018-10-09:
https://packages.gurobi.com/8.0/gurobi8.0.0_linux64.tar.gz is still there in the interim, they just do not advertise its existence.

jwnimmer-tri, 2018-10-27:
The URL above can be used as a work-around as long as Drake only supports 8.0.0.  Issue #9829 will be used to track Drake's upgrade to use a newer version, instead.

closed 2018-10-27 by jwnimmer-tri