RobotLocomotion/drake #9644: `documentation_pybind.h` is huge

EricCousineau-TRI, 2018-10-10:
At present, we dump all documentation bits into `documentation_pybind.h`.
It hurts CLion usage, as the current file is huge.

It shouldn't be too hard to have `mkdoc.py` generate different headers based on the directory (or namespace); it should also be easy to specify what directories we want to generate (to quell `genfiles`s requirement for outputs).

\cc @jwnimmer-tri @jamiesnape @amcastro-tri

jamiesnape, 2018-10-10:
I think that was the original plan, to be honest. Probably says more about CLion, than `documentation_pybind.h`, however.

jwnimmer-tri, 2018-10-10:
My vote would be for one header per top-level directory (attic, common, manipulation, systems, multibody, etc) to keep it simple for now (that set changes rarely -- the second-tier folders change more often).  And to be clear, still only run mkdoc once for all of libdrake, but have it emit a dozen'ish doc files.  That should mitigate IDE complaints and rebuild times -- bazel will not recompile uses of the header that is regenerated, as long as its contents stay the same.  If we find that the string tables are still too large, we can add in the second layer later on.

jamiesnape, 2018-10-12:
> My vote would be for one header per top-level directory [...]

Agreed.

EricCousineau-TRI, 2019-04-04:
I believe that #11131 may have slowed down `mkdoc` generation from ~50s to ~80s (on my beastly Puget); not a huge pain point, but given that all modules currently depend on this, it's a bit of a bottleneck that could be better alleviated by addressing this issue in the near-term.

Assigned `medium` priority for now.

EricCousineau-TRI, 2019-04-12:
Hm... On my laptop (8 cores), this is taking >5min to complete...

jamiesnape, 2019-04-12:
> 8 cores

It only uses one, if I am not mistaken.

EricCousineau-TRI, 2019-04-12:
That is correct; sorry, had posted number of cores to indicate possible resource starvation (not just intrinsic timing issues).

jwnimmer-tri, 2019-04-12:
My laptop is did not show that kind of slowdown.  I knew in #11131 I did some O(N) things that should be O(1) dicts; if it's a pain point I could amend that PR a little bit, before we fracture the header?

jamiesnape, 2019-04-12:
I'm not seeing it either, but it would be quicker to try rather than to split the file at this stage, I think.

EricCousineau-TRI, 2020-01-21:
Copying @RussTedrake's comment from Slack (not bumping priority yet, but noting the pain point):

> russtedrake Yesterday at 6:04 PM
> I happened to just catch a snapshot of how long it actually takes to compile the bindings on my mac  Action bindings/pydrake/documentation_pybind.h; 594s remote-cache, darwin-sandbox .  It's not my main workflow; I'm not complaining.  Just thought 10 min was impressive.  Especially since this was testing a change to a single variable name that amounted to < 20 characters.  :wink: 

[link](https://drakedevelopers.slack.com/archives/C0JNAGTNW/p1579561448000600)

EricCousineau-TRI, 2020-01-21:
That being said, @jwnimmer-tri, if you have time, could you point out which spot would need the modifications to be O(1)?

jwnimmer-tri, 2020-01-21:
I believe the slowdown introduced by #11131 was `extract_comment`'s search of `deprecations` for the `MACRO_INSTANTIATION` associated with the current comment.  But at this point I'd really just profile the python and see what the actual hot spot is.

ToffeeAlbina-TRI, 2020-04-09:
This should be worked on in combination with the bindings work
RobotLocomotion/drake#7889

EricCousineau-TRI, 2020-05-02:
Lower priority, as `attic` has been separated out.

jamiesnape, 2020-05-04:
Honestly, I would close this. It ends up being inseparable from #7889 and highly coupled to whatever the issue is for splitting libdrake.so (which I can't find at the moment).

EricCousineau-TRI, 2020-05-04:
SGTM. Will re-open if duration / lack of granularity pains us more in the future before #7889 lands.

closed 2020-05-04 by EricCousineau-TRI