RobotLocomotion/drake #9645: pydrake: Remove `pydrake.examples`

EricCousineau-TRI, 2018-10-10:
At present, some components from `//examples` are bound and exposed via `pydrake.examples`.

They should be deprecated, then removed.

\cc @RussTedrake @jwnimmer-tri

RussTedrake, 2018-10-10:
Tagging #9646 as relevant.

The other ideas/concerns that we discussed:
- [ ] Can consumers of the binary release of drake still run the examples?  I suspect that many users will try drake via binary first/only.  Letting them run examples seems valuable.  
- [ ] Can an example directory provide some form of installation?  I'm thinking more clearly now of an example subdir as an application of drake (not part of libdrake.so).  Like a separate repo that just happens to be collocated for convenience.  In that mindset, it seems reasonable for a separate app to have it's own install logic/artifacts?

jwnimmer-tri, 2018-10-10:
Neither #9646 nor this issue formulates the way that makes the more sense to me.  I will try to write up longer thoughts later, but to me the only important thing is whether the API to the artifact being shipped is via library calls (linking / loading) or via a binary program (new process).  I am more than happy to ship and an all _sealed_ binary demos -- where the stable interface to the user is the README.md and command-line arguments -- no matter where they live in the source tree.  When the API is via a library (be it C++ headers or python module), I think it should not live in examples.  In short, the only public targets of `//examples` (whether they be installed, or just consumed via drake_bazel_external) should be `foo_binary` or `filegroup`, not foo_library.  I think that will make it much easier to clarify which _library_ APIs are sensitive / stable.

jwnimmer-tri, 2018-10-10:
> Can an example directory provide some form of installation?

Per #4224, yes, the current intent is that all of the example applications are (or at least could be) installed.

closed 2024-03-21 by jwnimmer-tri