RobotLocomotion/drake #9646: manipulation_station: Decide on downstream consumption for manipulation class

EricCousineau-TRI, 2018-10-10:
My attempt to summarize the Slack discussions between (@RussTedrake and @jwnimmer-tri) and (@RussTedrake and myself):

1. We presently have `//examples/manipulation_station`, which is great for serving as a gallery example.
2. Russ would like Python bindings exposed for consumption in his [manipulation course](http://manipulation.csail.mit.edu/)
3. Per #9645, we shouldn't be exposing example / gallery code in `pydrake.examples`.
4. Russ would like a stable API for the class to consume as library code, and possibly utilities (such as systems to manipulate joints via a UI)
5. Some of the binaries may not be for external use (e.g. really just Drake demos). They should be runnable.
6. While this is more of an application than general library code, Russ has mentioned that developing this in the same code repository simplifies code development, specifically for driving API development.

@RussTedrake Please let me know if I missed anything here.

Here are some options that I can think of:

(1) Move all of `//examples/manipulation_station` into `//manipulation`, and make it supported API code, and expose the bindings via `pydrake.manipulation`

This would work, but starts to bloat Drake as a library with application code. For prior art, I'm not aware of anything within Drake that does this, aside from the fact that we include some specific models with Drake.

(2) Keep `//examples/manipulation_station` and expose the application-specific bits for downstream consumption, both C++ and Python. (e.g. `drake_manipulation_station` as a C++ module + Python package)

If we do this, then we should have similar tooling for packaging items for installation and testing them in this environment.

(3) Develop all of `manipulation_station` external to Drake.

We lose out on exhibiting this as an example, and add overhead for development process of the Manipulation class.

We do this presently for Anzu, but we have a team of TRI people working on this; there is overhead to drive Drake API changes, and it does require additional effort to port things back to Drake as library / example code. (More overhead for library code.)

\cc @jamiesnape

EricCousineau-TRI, 2018-10-10:
Within the short term, I'm fine with `pydrake.examples.manipulation_example` landing as-is to unblock development; that can be addressed as part of #9645, depending on how we decide to move forward here.

As an example, I'm going to submit a quick PR showing what it takes to create an Python example of `van_der_pol_oscillator` and install it under `share/drake/examples/`
(I will attempt to include docstring as well; we can figure out if we want Sphinx stuff when we do the gallery).

RussTedrake, 2018-10-10:
I also think that having this code inside drake provides important coverage for the manipulation code that is being developed in various repos.

I think I agree that (2) seems like a good outcome.

jwnimmer-tri, 2018-10-10:
(2) blurs the line for how sensitive Drake Developers need to be to API breaks.  I see no problem with (1).  There is plenty of relatively specific code installed by Drake in consumable library form.  I don't see what "bloat" is so worrying.  What is the acute problem with (1)?

jwnimmer-tri, 2018-10-11:
Is manipulation_station going to use PCL?  In general, I am not comfortable shipping a library that depends on PCL, no matter where it lives in git.

RussTedrake, 2018-10-11:
i was not proposing that the drake example touch PCL.  i want to put as many components as I can inside drake, but since I thought I cannot put them all in that I was thinking along the lines of "installing" the drake/examples/manipulation_station components as a library.

EricCousineau-TRI, 2018-10-12:
Per f2f with @jwnimmer-tri during platform meeting, it seems like it'd be best to use our current prior art:
- keep code in `//examples/manipulation_station`
- put Python bindings in `//bindings/pydrake/examples`, keeping an eye out for ODR issues

Once things settle after the course, we can resolve the issues per #9645 in a longer term and more though-out fashion.

Might that work?

RussTedrake, 2018-10-13:
Will only complain if i hit a blocker

closed 2018-11-25 by RussTedrake