RobotLocomotion/drake #9651: pydrake.math.RigidTransform sphinx does not show constructors

RussTedrake, 2018-10-11:
The documentation here:
http://drake.mit.edu/pydrake/pydrake.math.html?highlight=rigidtransform#pydrake.math.RigidTransform

does not capture the constructor bindings I see here:
https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/math_py.cc#L69

Not sure how wide spread the issue is.  Just wanted to capture the instance I observed.

RussTedrake, 2018-10-11:
A few others I've observed to be missing:
- Isometry3 constructors.
- LeafSystem._DeclareAbstractOutputPort

jamiesnape, 2018-10-11:
There is a Sphinx configuration option. By default constructors are not documented separately. I will look into changing the option.

RussTedrake, 2018-10-11:
Thanks.  I don’t see how that could have address the leafsystem bindings, though.

jamiesnape, 2018-10-11:
So this issue is not just about constructors? Can you update the title/description?

EricCousineau-TRI, 2018-10-12:
Because we decided to use private names for overloaded methods, `_MethodName`, we run into issues like this.

My preference is that we only allow certain methods in; we may be able to work this into our current `sphinx_pydrake` plugin.

jamiesnape, 2018-10-12:
There is a Sphinx setting for private members to be documented. Either we set that, change our naming convention, or accept that these methods are not going to appear in the online docs. Adding the plugin should be a last resort. We should not overcomplicate this.

EricCousineau-TRI, 2018-10-15:
I don't want private members documented unless they're intended to be used by users.

I would like to know what other projects do when they have private / protected methods to be overridden; if they do not prefix with `_`, then we shouldn't.
If they do, then we should target specific members.

sherm1, 2018-12-05:
I just re-discovered this problem in #10136 when I added some new LeafSystem method (DeclareDiscreteState). Most of LeafSystem's useful content is protected.

jamiesnape, 2018-12-05:
@sherm1, therefore the useful content in LeafSystem was not documented online, or it affected you in #10136 in some other way?

sherm1, 2018-12-05:
Right, there was no documentation online for the new methods I added, even though they had elaborate doxygen in C++. I'm not blocked by it, but it seems like a major gap in the Python docs. Those protected methods are really necessary for Python users to make Drake Systems.

PR #9652 (2018-10-11): Render __init__ docstrings in pydrake Sphinx documentation
Fixes #9651.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9652)
PR #11073 (2019-04-01): py systems: Use `{NAME}` for protected methods; deprecate `_{NAME}`
Resolves #9651

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11073)
closed 2019-04-01 by EricCousineau-TRI