RobotLocomotion/drake #9661: fix inconsistent use of `timestep` and `time_step` throughout the codebase

RussTedrake, 2018-10-12:
I think different authors have used different variants.  I don't have a strong opinion, but would like things to be consistent.

jamiesnape, 2018-10-12:
Almost raised this in https://github.com/RobotLocomotion/drake/pull/9200. `timestep` is not in the dictionary, of course.

jwnimmer-tri, 2022-05-01:
The two-word is used approx 900 times, the one-word approx 300 times.  The 900-times is mostly dynamics code (multibody, simulator), the 300-times is mostly Russ code (traj opt).

Shall we settle on the two-word version for new code?

Is it worth fixing the one-word version in old code (possibly only non-API-changing locations, to avoid churn)?

RussTedrake, 2022-05-01:
I'm happy to go with the two-word version, and even with changing the API at some point if we decide that's best.  I agree that there is significant value in consistency.

sherm1, 2022-05-02:
Grammarhow favors "time step": https://grammarhow.com/timestep-time-step-or-time-step/

PR #19478 (2023-05-24): Rename timestep => time_step in remaining uses/comments throughout
Resolves #9661.

+@sherm1 for both reviews, please.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19478)
commit fa6a45ed (2023-05-24): Rename timestep => time_step in remaining uses/comments throughout (#19478)

Resolves #9661.
closed 2023-05-24 by RussTedrake