RobotLocomotion/drake #9666: MBP + SceneGraph support clone or temporary modifications.

siyuanfeng-tri, 2018-10-12:
A common use case for planning is to take a given model of the world, make some local changes to it, do real work, then throw those modifications away. It would be nice to support this work flow. 
A easy and safe approach is to support cloning of the dual, and throw away the copy when the planner is done. 
Another more complicated but could be more performant way is to keep the const original model, and add some mutable representation on top of it.

amcastro-tri, 2018-10-12:
Relates to workflow depicted in #9665. For as long we resolve #9665, we can work around not having this capability. However that workflow would require keeping track of what we add/delete from our world.

EricCousineau-TRI, 2020-04-30:
This seems like it can be covered by #12703 (if that issue can be handled by multiple applications).

closed 2020-04-30 by EricCousineau-TRI