RobotLocomotion/drake #9669: Context should fail-fast when FixInput is given a mis-typed Value

jwnimmer-tri, 2018-10-14:
Inspired by #9591.  An input port was Fixed with a wrongly-sized BasicVector.  The error message was an obscure DRAKE_DEMAND without much context on the wrong sizes or the System in question.  That PR proposes to improve failure text at the DEMAND size.  My counter-proposal is that when the user calls FixInputPort with a value that could never possibly work, we should error out then, instead of during System evaluation.

jwnimmer-tri, 2018-10-14:
In pursuing this, for fixing abstract inputs, I was going to rely on the declared type of the port to compare the Fixed value against.

Per this block from system.h:
https://github.com/RobotLocomotion/drake/blob/de81cd2f9c43152505b80754822b29cde039fc3b/systems/framework/system.h#L121-L132

And this block from leaf_system.h:
https://github.com/RobotLocomotion/drake/blob/de81cd2f9c43152505b80754822b29cde039fc3b/systems/framework/leaf_system.h#L418-L438

... when the user declares the abstract input port, they should either provide a model_value or override some method.  As it turns out, once we start relying on this, we find many place in the code that failed to so do.  I will plan to fix those up first.

jwnimmer-tri, 2018-10-14:
The other question is for vector (i.e., non-abstract) input ports, if we want to enforce that the BasicVector subtype is Fixed by the user, or if we magically promote the untyped vector to the correct type, as Fix-time.  I'll elaborate on that separately, once I have some examples ready.

jwnimmer-tri, 2018-11-06:
After that PR train, the only remainaing unmerged fixes are the WIP commits in #9620.  They relate to whether and how FixInputPort for inputs declared with a BasicVector subtype should operate.  I'm going to back burner that a little bit for now.  I think the right path forward will be to sketch up the two design choices into separate PRs, and then see which design consequence we like best.

sherm1, 2020-06-02:
@jwnimmer-tri should this issue still be open? If so maybe replace with a narrower one?

jwnimmer-tri, 2020-06-02:
The additional checking mentioned above is not yet implemented; the WIP branch is https://github.com/jwnimmer-tri/drake/tree/fix-error (and some of those PRs linked about).  I am content to keep using this issue as the discussion point and keep it open.

I have lowered the priority to `low` though -- the size mismatch was probably a bigger headache than the subtype checking.

closed 2024-11-01 by jwnimmer-tri