RobotLocomotion/drake #9683: Broken links on drake.mit.edu

jamiesnape, 2018-10-16:
#9681 reminds that we have multiple broken links in the documentation.

Links to (at least) the following targets are broken:

| Target                                                                                               | Source                                                                                      |
|------------------------------------------------------------------------------------------------------|---------------------------------------------------------------------------------------------|
| https://drake.mit.edu/doxygen_cxx/stochastic_systems.html                                            | https://drake.mit.edu/                                                                      |
| https://github.com/RobotLocomotion/drake/tree/master/drake/examples                                  | https://drake.mit.edu/                                                                      |
| https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_direct_trajectory_optimization.html      | https://drake.mit.edu/                                                                      |
| https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_inverse_dynamics_controller.html         | https://drake.mit.edu/                                                                      |
| http://www.doxygen.org/index.html                                                                    | https://drake.mit.edu/doxygen_cxx/group__systems.html                                       |
| https://simtk.org/api_docs/simbody/latest/classSimTK_1_1Force_1_1MobilityLinearStop.html             | https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_rigid_body_plant.html           |
| https://en.wikipedia.org/wiki/Monomial_order,                                                        | https://drake.mit.edu/doxygen_cxx/structdrake_1_1symbolic_1_1_graded_reverse_lex_order.html |
| https://inst.eecs.berkeley.edu/~ee127a/book/login/l_socp_soc.html                                    | https://drake.mit.edu/doxygen_cxx/classdrake_1_1solvers_1_1_lorentz_cone_constraint.html    |
| https://drake.mit.edu/doxygen_cxx/a,%20b,%20...%20x,%20y,%20...                                      | https://drake.mit.edu/doxygen_cxx/classdrake_1_1solvers_1_1_system_identification.html      |
| https://en.wikipedia.org/wiki/Richardson27s_theorem                                                  | https://drake.mit.edu/pydrake/pydrake.symbolic.html                                         |
| https://drake.mit.edu/doxygen_cxx/python_bindings.html                                               | https://drake.mit.edu/python_bindings.html                                                  |
| https://github.com/googlei18n/noto-fonts/blob/master/hinted/NotoSans-Regular.ttc?raw=true            | https://drake.mit.edu/unicode_tips_tricks.html                                              |
| https://en.cppreference.com/w/cpp/concept/RandomNumberDistribution                                   | https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1internal_1_1_random_source.html  |
| https://drake.mit.edu/doxygen_cxx/namespacestd.html                                                  | https://drake.mit.edu/doxygen_cxx/diagram_8h_source.html                                    |
| https://www.math.uh.edu/~jingqiu/math4364/spline.pdf,                                                | https://drake.mit.edu/doxygen_cxx/classdrake_1_1trajectories_1_1_piecewise_polynomial.html  |
| https://en.wikipedia.org/wiki/Spline_interpolation,                                                  | https://drake.mit.edu/doxygen_cxx/classdrake_1_1trajectories_1_1_piecewise_polynomial.html  |
| https://drake.mit.edu/doxygen_cxx/x,%20y...                                                          | https://drake.mit.edu/doxygen_cxx/namespacedrake_1_1solvers.html                            |
| http://www.generic-programming.org/languages/cpp/techniques.php                                      | https://drake.mit.edu/doxygen_cxx/group__templates.html                                     |
| https://en.wikipedia.org/wiki/Rodrigues27_rotation_formula                                           | https://drake.mit.edu/pydrake/pydrake.multibody.inverse_kinematics.html                     |
| http://www.coin                                                                                      | https://drake.mit.edu/pydrake/pydrake.solvers.mathematicalprogram.html                      |
| https://github.com/google/googletest/blob/master/googletest/docs/AdvancedGuide.md                    | https://drake.mit.edu/styleguide/cppguide.html                                              |
| https://en.cppreference.com/w/cpp/concept/PODType                                                    | https://drake.mit.edu/styleguide/cppguide.html                                              |
| https://github.com/RobotLocomotion/drake/blob/master/multibody/collision/test/multi_model_groups.sdf | https://drake.mit.edu/doxygen_cxx/group__collision__filter__group.html                      |

mpetersen94, 2020-04-29:
How was this list generated? I think a bunch of these have been repaired since this issue was posted.

jamiesnape, 2020-04-29:
`grep` and `linkchecker` on Ubuntu. This is probably more up to date as far as the whole repo goes :
```
http://ccom.ucsd.edu/~optimizers/usage/c/
http://continuousphysics.com/Bullet/phpBB2/viewtopic.php?t=565
http://docs.opencv.org/2.4/modules/calib3d/doc/
http://home.uchicago.edu/~sctchoi/courses/cs138/interp.pdf
http://math2.uncc.edu/~shaodeng/TEACHING/math5172/Lectures/Lect_15.PDF
http://paulbourke.net/geometry/superellipse/
http://people.sc.fsu.edu/~burkardt/m_src/lemke/lemke.m
http://sodipodi.sourceforge.net/DTD/sodipodi-0.dtd
http://www.coin-or.org/Ipopt/documentation/node23.html
http://www.coin-or.org/Ipopt/documentation/node38.html
http://www.coin-or.org/Ipopt/documentation/node40.html
http://www.cs.cmu.edu/afs/cs/academic/class/16741-s07/www/Lecture8.pdf
http://www.dm.uniba.it/~testset
http://www.engr.iupui.edu/~skoskie/ECE684/Riccati_algorithms.pdf
http://www.generic-programming.org/languages/cpp/techniques.php
http://www.greentoken.de/onlineconv/
http://www.jidesoft.com/products/oss.htm
http://www.stack.nl/~dimitri/doxygen
https://blobscdn.gitbook.com/v0/b/gitbook-28427.appspot.com/o/assets%2F-LgYfwVg89cfloOSocyC%2F-LhJQm219Jka_jubcY86%2F-LhJQsQLlSIty5iywZxX%2FUntitled.gif?alt=media&token=c343b42a-5927-48c9-981b-b2074ae3da56
https://blobscdn.gitbook.com/v0/b/gitbook-28427.appspot.com/o/assets%2F-LgYfwVg89cfloOSocyC%2F-Lhq9MPiXqZNYqPUQ9bG%2F-Lhq9Tp8S5LlQUyPQaW-%2Fcart_pole_tracking.gif?alt=media&token=d5e653f0-810e-4008-8279-f1607cb12664
https://cdnjs.cloudflare.com/ajax/libs/mathjax/2.7.5
https://courses.cit.cornell.edu/mae3050/mae3050ThinAirfoils.pdf
https://docs.bazel.build/versions/master/skylark/cookbook.html
https://drake.mit.edu/cxx_inl.html
https://drake.mit.edu/doxygen_cxx/html/group__mesh__intersection__benchmarks.html
https://drake.mit.edu/doxygen_cxx/html/group__render__engine__benchmarks.html
https://inst.eecs.berkeley.edu/~ee127a/book/login/l_socp_soc.html
https://jide-oss.dev.java.net/servlets/ProjectDocumentList
https://manipulation.csail.mit.edu
https://people.sc.fsu.edu/~jburkardt/f_src/triangle_dunavant_rule/triangle_dunavant_rule_prb_output.txt
https://simtk.org/api_docs/simbody/latest/classSimTK_1_1Force_1_1MobilityLinearStop.html
https://www.coin-or.org/Ipopt/documentation/node40.html
https://www.coin-or.org/Ipopt/documentation/node42.html
https://www.kuka.com/-/media/kuka-downloads/imported/48ec812b1b2947898ac2598aff70abc0/spez_lbr_iiwa_en.pdf
https://www.stack.nl/~dimitri/doxygen/
https://www.stack.nl/~dimitri/doxygen/manual/docblocks.html
https://www.wolframcloud.com/objects/60e3d6c1-156f-4604-bf70-1e374f47140
```

jwnimmer-tri, 2023-01-27:
After https://github.com/RobotLocomotion/drake/pull/18679 lands, these are the remaining three links worth worrying about:

From drake/common/trajectories/piecewise_polynomial.h:
```
http://home.uchicago.edu/~sctchoi/courses/cs138/interp.pdf
https://www.math.uh.edu/~jingqiu/math4364/spline.pdf
```

Added in #4334 and #9150.

From drake/math/random_rotation.h:
```
http://planning.cs.uiuc.edu/node198.html
```

Added in #12562.

jwnimmer-tri, 2023-01-27:
I'll assign this over to @hongkai-dai for a decision.  See the immediately prior comment.  This is low priority.

Would you like to try to dig up some replacement citations for those three links?  If yes, you could make a PR with the new ones.

If not, feel free to just close the issue.

closed 2025-11-17 by jwnimmer-tri