RobotLocomotion/drake #9688: MultibodyPlant: Define scalar conversion workflow with Scene Graph?

EricCousineau-TRI, 2018-10-16:
Is there a workflow already in mind for doing scalar conversion for MBP with a `SceneGraph`?

At present, `MultibodyPlant<T>::MultibodyPlant<U>` copies geometry registration information (`source_id_`, etc.), but does not try to ensure that MBP has a valid reference to an equivalent `SceneGraph<U>`, so post-conversion, we have `geometry_source_is_registered() == true` while `scene_graph_ == nullptr`, which has a funny smell to it.

How should a user supply the converted `SceneGraph`? Can we update `multibody_plant_test` to use that workflow?

---

We ran into this post-merge of #9671, which is temporarily fixable by reverting the offending error (indicated by ASan):
`runtime error: member call on null pointer of type 'drake::geometry::SceneGraph<Eigen::AutoDiffScalar<Eigen::VectorXd> >'`
Offending Source: https://github.com/RobotLocomotion/drake/blob/167f7dd/multibody/multibody_tree/multibody_plant/multibody_plant.cc#L1617

\cc @jwnimmer-tri @amcastro-tri

jwnimmer-tri, 2018-10-16:
The SceneGraph pointer is only transient anyway (during configuration).  Once the MBP is finalized, it is reset to nullptr.  As long as you don't want to unfinalize after transmogrification, I think we're okay.

EricCousineau-TRI, 2018-10-16:
Gotcha. So as stated, this is not an issue, but will become an issue if we haven't decoupled the default allocation of a `QueryObject` from `SceneGraph` once we're required to allocate model values (when you remove that overload)?

jwnimmer-tri, 2018-10-16:
Yes, I think that's accurate.  I believe the plan anyway for QueryObject is to make it independently constructable, so we don't need a SG to make one.  I'll have to solve it one way or another when I remove (deprecate) the DeclareAbstractInputPort overloads that lack model values (#9669).

EricCousineau-TRI, 2018-10-16:
Alrighty, sounds good. Will close this issue, then. Thanks!

closed 2018-10-16 by EricCousineau-TRI