RobotLocomotion/drake #9691: Need for a composite MBP + SG

amcastro-tri, 2018-10-17:
We had several talks about this. #9665 is the most recent discussion that references this design.
The ability to easily merge MBP with SG is blocking in that many of our new IK and planning tools really need both in sync, with the ability to make changes and with simple APIs to perform both geometric queries and multibody queries. The ability to seamlessly work with a single context strongly relates to this issue.

cc'ing @hongkai-dai, @avalenzu, @siyuanfeng-tri

edrumwri, 2018-10-17:
FYI: I was working on this yesterday. I hope to have a prototype completed
today. Any interested testers please ping me.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

EricCousineau-TRI, 2018-10-22:
Per @amcastro-tri's comment, motivating use case relevant to Drake is to enable `InverseKinematics` to easily pose geometric-based constraints, such as the MBP analog of `MinDistanceConstraint`, and figure out how that affects the current API:

https://github.com/RobotLocomotion/drake/blob/d2a20870979d1991754528b18c20c2ba9130167a/multibody/inverse_kinematics/inverse_kinematics.h#L29-L30

amcastro-tri, 2018-10-23:
To be more specific yet. We want to be able to implement this [single line of code](https://github.com/RobotLocomotion/drake/blob/master/attic/multibody/rigid_body_constraint.cc#L1957) from the previous IK solver in the new `drake::multibody::InverseKinematics`. @hongkai-dai wrote this so I am sure he can give very precise details locally.

**That defines** the problem we want to solve.

A possible solution we talked about in Ann Arbor was this "composite MBP+SG diagram". However other solutions might be possible and I believe we should consider them.

edrumwri, 2019-11-22:
I believe this has been resolved. Please reopen if you feel I am mistaken.

closed 2019-11-22 by edrumwri