RobotLocomotion/drake #9702: Event handling is not triggered on systems that require discrete or unrestricted update handling at time zero

edrumwri, 2018-10-18:
(and documentation no longer indicates that we require CalcNextUpdateTime() to return a time strictly in the future). Proposed fix is to do any such updates in Simulator::Initialize().

EricCousineau-TRI, 2018-10-24:
I believe this may be a small reproduction case:
```python
from pydrake.all import DiagramBuilder, AbstractValue, LeafSystem, ZeroOrderHold, Simulator

class Source(LeafSystem):
    def __init__(self):
        LeafSystem.__init__(self)
        self.output = self._DeclareAbstractOutputPort(
            "output", alloc=lambda: AbstractValue.Make(""), calc=self._Output)

    def _Output(self, context, raw):
        raw.set_value("u[t={}]".format(context.get_time()))

class Stuff(LeafSystem):
    def __init__(self, name, dt):
        LeafSystem.__init__(self)
        self.input = self._DeclareAbstractInputPort("input")
        self._DeclarePeriodicPublish(dt)
        self.name = name
    
    def _DoPublish(self, context, events):
        u = self.EvalAbstractInput(context, self.input.get_index()).get_value()
        t = context.get_time()
        print("{}: t={:.3g}, y=({})".format(self.name, t, u))

builder = DiagramBuilder()
source = builder.AddSystem(Source())

dt = 1
stuff = builder.AddSystem(Stuff("direct", dt))
builder.Connect(source.output, stuff.input)

stuff_zoh = builder.AddSystem(Stuff("zoh   ", dt))
zoh = builder.AddSystem(ZeroOrderHold(dt, AbstractValue.Make("")))
builder.Connect(source.output, zoh.get_input_port(0))
builder.Connect(zoh.get_output_port(0), stuff_zoh.input)

diag = builder.Build()

sim = Simulator(diag)
sim.set_publish_every_time_step(False)
sim.Initialize()
sim.StepTo(3)
```

What it prints out:
```
direct: t=0, y=(u[t=0.0])
zoh   : t=0, y=()
direct: t=1, y=(u[t=1.0])
zoh   : t=1, y=()
direct: t=2, y=(u[t=2.0])
zoh   : t=2, y=(u[t=1.0])
direct: t=3, y=(u[t=3.0])
zoh   : t=3, y=(u[t=2.0])
```

What it *should* print out:
```
direct: t=0, y=(u[t=0.0])
zoh   : t=0, y=()
direct: t=1, y=(u[t=1.0])
zoh   : t=1, y=(u[t=0.0])  <-- Missing from above
direct: t=2, y=(u[t=2.0])
zoh   : t=2, y=(u[t=1.0])
direct: t=3, y=(u[t=3.0])
zoh   : t=3, y=(u[t=2.0])
```

Ran into this when talking with @thduynguyen, relevant to @SeanCurtis-TRI's PR on an image writer.

sherm1, 2018-11-30:
#9707 duplicates this and has more information so I'll close this one.

closed 2018-11-30 by sherm1