RobotLocomotion/drake #9707: DeclarePeriodicUpdate with offset=0 does not generate an update at time=0.

RussTedrake, 2018-10-19:
Getting the DiscreteDerivative System implemented is continuing to test the subtler points of discrete/continuous time interaction.  I think that right now it's close, but still missing the first update.  

The system in #9533 `discrete_derivative.h` is 
```
/// This is implemented as the linear system
///   @f[ x₀[n+1] = u[n], @f]
///   @f[ x₁[n+1] = x₀[n] = u[n-1], @f]
///   @f[ y(t) = (x₀[n]-x₁[n])/h. @f]
```

I've added some debugging text (which I've pushed, in case it's helpful), which generates the following output when running `discrete_derivative_test.cc`:
```
t = t, y = ((xd0_0 - xd0_2) / 0.10000000000000001)                                       ((xd0_1 - xd0_3) / 0.10000000000000001)
t = t, x = xd0_0 xd0_1 xd0_2 xd0_3, x+ =  u0_0  u0_1 xd0_0 xd0_1
t = 0, y = 0 0
t = 0.0306091, y = 0 0
t = 0.0612182, y = 0 0
t = 0.0918274, y = 0 0
t = 0.1, x = 0 0 0 0, x+ = 4.2 5.3   0   0
t = 0.122436, y = 42 53
t = 0.153046, y = 42 53
t = 0.183655, y = 42 53
t = 0.2, x = 4.2 5.3   0   0, x+ = 4.4 5.6 4.2 5.3
t = 0.214264, y = 2 3
t = 0.244873, y = 2 3
t = 0.275482, y = 2 3
t = 0.3, x = 4.4 5.6 4.2 5.3, x+ = 4.6 5.9 4.4 5.6
t = 0.306091, y = 2 3
t = 0.3367, y = 2 3
t = 0.367309, y = 2 3
t = 0.397919, y = 2 3
...
```

The first time that the discrete derivatives are called is after the first period.  Consider the discrete-time system:
```
x[n+1] = f(x[n], u[n])
y[n] = g(x[n],u[n])
```
If I simulate this system in drake and look at the outputs only at times `t` that are a multiple of the periodic sampling time, `h`, then I should be able to see a "perfect" reconstruction:
```
x(t+h) = f( x(t), u(t) )
y(t) = g( x(t), u(t) )
```
Notably, `x(h) = f( x(0), u(0) )`.  That is currently not happening.  In the system above, the inputs are never being evaluated until `u(h)`.

In our implementation, which is now closer to being correct for publishing the outputs (thank you), we have the additional detail that `y(t) = y(n*h)` for all `(n-1)*h < t <= n*h` -- which feels a little strange but I think is ok; that part is now correct.  This is because the same is true for `x(t)` -- the value for `x(a+h)` has to be computed at time `a`, and so `x` is given that value throughout the interval `(a, a+h]`.  Do we all agree?

My understanding is that we're missing the first publish, then things should be correct.

RussTedrake, 2018-10-19:
I think we need to have a basic discrete time system (with control inputs)... like `x[n+1] = x[n] + u[n]` with the closed-form solution (e.g. the impulse response should be the unit step) implemented as a unit test.  I'm a bit embarrassed that we don't seem to.  We could have an input that literally evaluates `u=1` iff `t==0` for the impulse.  That would have caught this (`x` would have stayed at the initial conditions always, and never evaluated `u` at time zero).  

cc @mitiguy .

RussTedrake, 2018-10-19:
That test really belongs in `integrator_test.cc`.  Can't believe that it wasn't there.

edrumwri, 2018-10-25:
FYI @RussTedrake I'm in progress on this issue. We'll get some more unit tests in place for discrete system during resolution.

edrumwri, 2018-10-27:
@RussTedrake looking at your example with the Heaviside step function:

> I think we need to have a basic discrete time system (with control inputs)... like x[n+1] = x[n] + u[n] with the closed-form solution (e.g. the impulse response should be the unit step) implemented [...]

I want to make sure that you agree with my interpretation as I'm building unit tests. My research indicates that it's equally valid to define the Heaviside function as both:

```
H(t) = { 0  for t < 0 and
           { 1  for t >= 0
```

and as:

```
H(t) = { 0  for t <= 0 and
           { 1  for t > 0
```

Since the discrete change happens instantaneously, it's also possible that H(t) takes on two values for any one `t`, further complicating the definition. For Simulator, that means that two values could be published at any point in time.

Now if we consider a different function: `x[n+1] = x[n] + 1` which we simulate using a hybrid continuous/discrete simulator from initial conditions x[0] = x(0) = 0, I believe all of the following behavior is correct (as long as it's consistent): 

(a) x(0 ≤ t < h) = 0, x(h) = 1
(b) x(0 ≤ t ≤ h) = 0, x(h) = 1
(c) x(0 < t < h) = 0, x(h) = 1
(d) x(0 < t ≤ h) = 0, x(h) = 1

Do you agree?

RussTedrake, 2018-10-27:
There are, in my opinion, standard methodologies for converting between discrete and continuous time, with the central concept being the transformation of the discrete-time unit impulse into continuous time as a box with area 1 taken to the limit where the width goes to zero (to obtain the continuous time impulse).  i agree that there is some ambiguity on whether that box is centered at zero or shifted, e.g. `u(t) = 1/h` for `[-h/2, h]` vs `[0, h]`, with the limit as `h => 0`.  The unit step is the integral of this system, which can lead to the variations you call out above for the Heaviside function.

But I *think* there is a correct answer for our case, which I tried to work out above.  The feature that is most important to me is that if I have a discrete-time system, e.g.`x[n+1] = x[n] + u[n]`, then looking at only the discrete sample times defined by that system's periodic update, I should get consistent behavior.  In this case, if `u[n] = 0` for all `n != 0, u[0] = 1`, then the output at the sample points should be `x[n] = 0` for `n < 1, x[n] = 1` for `n >= 1`.  Because of the nature of our integration, which only evaluates the inputs at the current time (as it should), then the output `y(t) = x(t)` should evaluate to `y(t) = 0` for `t<=0, y(t) = 1` for `t > 0`.  

So I think the right answer for your system above, `x[n+1] = x[n] + 1, x[0] = x(0) = 0`, is (e) none of the above.  I would think that the *only* correct answer for us is:
```
  x(0) = 0, x(0 < t <= h) = 1, x(h < t <= 2*h) = 2, ...
```
etc.  Publishing two values at any one time is definitely *not* the requested solution.

edrumwri, 2018-10-27:
Thank you for your detailed thoughts. Neither I nor the folks I've been discussing this with at HQ  are particularly confident with discrete systems. I'm going to ask for elaboration on some of your statements. The main thing I want clarity on is what behavior is necessary for correctness vs. what behavior you expect (perhaps driven by your experience with Simulink). I'm getting some mixed messages in that respect (due to ambiguity of words like "correct" and "right") with your statements:

> But I think there is a correct answer for our case ... The feature that is most important to me is that if I have a discrete-time system, e.g.x[n+1] = x[n] + u[n], then looking at only the discrete sample times defined by that system's periodic update, I should get consistent behavior.

You mean only that we can test whether the behavior that we have defined is consistent, right? I believe we're in agreement that a hybrid simulator can output one of multiple samples for a given time that are all consistent with theory. The only inviolable property for mathematical correctness seems to be that the integral of `x(t)` over interval of length `h` and centered at `h*n` is equal to `h*x[n]` as `h -> 0`. 

When you say:

> So I think the right answer for your system above, `x[n+1] = x[n] + 1, x[0] = x(0) = 0`, is (e) none of the above.

Just to confirm: this is only the behavior you'd like to see, not a statement on what is mathematically "right"?

RussTedrake, 2018-10-27:
I think we can check for correctness of a simulation of a discrete-time system with a periodic update, modeled conceptually as x[n+1] = f(x[n], u[n]).  In my mind, the "correctness" requires only that at the time intervals, t = n*h, the simulated output matches the math.  Between time-samples, we have more room for interpreting the discrete-time / continuous-time relationship.. although there is precident (e.g. in textbooks, and in Simulink).

The answer that I believe we should implement ("like to see"?) is what i proposed above, with ```x(0) = 0, x(0 < t <= h) = 1, x(h < t <= 2*h) = 2, ...``` is because of my thinking about how the rest of the simulation infrastructure is implemented.  I believe that, to be consistent with the discrete-time model, u[n] must be evaluated at u(n*h).  Since we can only evaluate inputs at the current context, the natural implication should be that we compute x[n+1] when the simulator is at time n*h.  Because it is consistent with the theory, and significantly easier to implement that something which tries to remember some updates and apply them at a later time, I think applying that update "immediately" -- so that evaluations by a continuous time system at times n*h < t <= n*(h+1) all take that value, is the solution that I would advocate for.  I also think it is mathematically consistent/satisfying with a few quick though experiments that i did in formulating this opinion.

edrumwri, 2018-10-30:
Your proposed solution:

>  I think applying that update "immediately" -- so that evaluations by a continuous time system at times nh < t <= n(h+1) all take that value

 initially sounded reasonable to me as well, but I've identified a particular flaw. The discrete variables are updated only through the event mechanism, whereas continuous variables (including time) are updated only through integration. That detail implies that discrete variables should only be updated at the end of a continuous time interval; if instead discrete variables are the ones updated at the end of the process, then the continuous time can become out of sync with the discrete variables.

From my example system, x[n+1] = x[n] + 1, x[0] = 0, stepping from time zero to unity would look like:

1. update discrete variables x[1] = x[0] + 1
2. integrate time to the next event t = 1

under your proposal which would leave us with an event (the next discrete update) in the queue. If we handle that event (which we should before Simulator "signs off" on the state, since events- whether those used for updating discrete state or otherwise- are treated in a uniform manner), we would then have x[2] = x[1] + 1 _but with `t` remaining at 1_. Alternatively, we can handle these particular events using special logic, which I think all stakeholders would find inelegant. 

@RussTedrake I advocate for a different approach: updating discrete variables using `DeclarePeriodicDiscreteUpdate(period, period)`. In other words, the first update should happen at the end of the first continuous time interval. 

I've identified two downsides:
1. We're currently able to transform time to iterate by simply dividing `context.get_time()` by `h`. The iterate would have to be offset by 1. This could be sugared by adding a GetPeriodicDiscreteUpdateIterate() method to LeafSystem, which would be helpful to have anyway.
2. Codebase changes can't simply be limited to `Simulator` (follows from 1.)

edrumwri, 2018-10-30:
On further reflection, let me try to address it the way you want to through the existing architecture, since I don't find your request unreasonable. We can revisit the above discussion if there appears to be no way around the software pain.

sherm1, 2018-10-30:
I'm not getting this yet -- let's discuss f2f w/whiteboard.

jwnimmer-tri, 2018-12-03:
PR #9766 (as auto-linked into the timeline about two posts prior) is under review.

PR #9766 (2018-12-14): Handles events at initial time step.
Resolves #9707 by:

1) Allowing timed events to occur at time zero.

2) Adding unit tests for simple discrete and hybrid systems including those suggested in #9707 (into simulator_test).

3) Adding a front-page Drake module explaining how discrete systems are modeled and simulated within Drake's hybrid systems framework.

4) Adding a detailed exposition to Simulator class documentation on the inner workings of `StepTo()` and `Initialize()` with respect to Publish and Update timing.

5) Adding `examples/fibonacci` to show a classic difference equation genera [... cut]
closed 2018-12-14 by edrumwri