RobotLocomotion/drake #9709: ManipulationStation should use actual workstation .sdf (and objects) instead of table.sdf

RussTedrake, 2018-10-19:
Decided to host the geometry (which might change at the solidworks level) in RobotLocomotion/models . Resolving this issue requires pulling in the data from that repo. @jwnimmer-tri has already given me some close-to-ready bazel foo to work from.

RussTedrake, 2018-10-20:
I've got the drake bazel scripts reading model files from the RobotLocomotion/models repo now.  But the station.sdf file does not appear to load correctly in drake_visualizer.  

Hoping @pangtao22 and/or @Sproulx can help me fix it.

RussTedrake, 2018-10-21:
Excellent!  Will pull it into my dev branch first thing tomorrow!

RussTedrake, 2018-10-22:
it also fails when trying to run from bazel-bin directly.

```
russt@russt-ThinkPad-P51:~/drake/examples/manipulation_station$ ~/drake/bazel-bin/examples/manipulation_station/mock_station_simulation
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not find resource: drake/external/models_robotlocomotion/manipulation_station/amazon_table_simplified.sdf
Aborted (core dumped)
```

jwnimmer-tri, 2018-10-22:
@RussTedrake Where is this code posted?  `drake::FindResourceOrThrow` is only for Drake's own models.  It is not built to locate workspace externals.  When models are declared as `data = []` dependencies in some rule, they will be found at something like `./external/robotlocomotion_models/dir/file.sdf` when run in the sandbox.  To avoid hard-coding paths like that, we should probably be using the https://github.com/bazelbuild/bazel/blob/master/tools/cpp/runfiles/runfiles_src.h component to interrogate the runfiles.  We don't have any examples of that yet to my knowledge (relates #9061).

RussTedrake, 2018-10-22:
https://github.com/RussTedrake/drake/tree/lab1

with calls, specifically at e.g. 
https://github.com/RussTedrake/drake/blob/lab1/examples/manipulation_station/manipulation_station.cc#L114

jwnimmer-tri, 2018-10-22:
Off the top of my head, I think
```cxx
const std::string table_sdf_path = 
    "external/models_robotlocomotion/manipulation_station/"
    "amazon_table_simplified.sdf";
```
would be right.

closed 2018-10-28 by RussTedrake