RobotLocomotion/drake #9711: Installation Issue

dabpanda, 2018-10-19:
While trying to install Drake on Ubuntu 16.04, I followed the documentation available in [official site](http://drake.mit.edu/python_bindings.html#using-python-bindings) but am stuck with an error in the `cmake -DWITH_GUROBI=ON -DWITH_MOSEK=ON ../drake`  step. 
[image]

hongkai-dai, 2018-10-19:
Which version of Mosek and Gurobi do you have on your machine?

EricCousineau-TRI, 2018-12-10:
Howdy @rahulvigneswaran! I'm not sure if you've yet had a chance to revisit this issue. If you want, you can see the logic for how CMake will try to find the location of Gurobi in known locations:
https://github.com/RobotLocomotion/drake/blob/94bb5e4d1626dc3d5d983a7835b7047b50dcd1c6/cmake/modules/FindGurobi.cmake#L4-L11

I'm going to close this for now.
Please do let us know if you still need help, in which case we can re-open the issue.

closed 2018-12-10 by EricCousineau-TRI