RobotLocomotion/drake #9720: Add exclusion plane logic to manipulation_station simulation?

RussTedrake, 2018-10-19:
We've implemented non-collision constraints at the firmware level. Would be nice to have them in sim, too (so that we can understand that, e.g., the system will shut down if you move the elbow too close to the truss)

RussTedrake, 2018-11-11:
Should also alarm on driver-level faults:
- [ ] torque limits

(are there any others?  @pangtao22?  @siyuanfeng-tri ?)

pangtao22, 2018-11-11:
A Cartesian speed limit for all joint centers is programmed at the firmware level too. 

In addition, the robot also shuts down if the commanded position is “too far away” from its current position. I’m not sure if it is in the iiwa driver or one of the few firmware safety functions that cannot be turned off. 

If we’d like to accurately reproduce the collision checking that the firmware is performing, we’ll need to use the spherical collision geometry defined in one of iiwa’s manuals.

RussTedrake, 2019-05-24:
WON'T FIX.  (it could be an issue to track for class, if we need it for the fall)

closed 2019-05-24 by RussTedrake