RobotLocomotion/drake #9727: refactor StationSimulation components into iiwa directory.

RussTedrake, 2018-10-19:
likely consolidate kuka controller implementations to KukaJointImpedanceControl, KukaTorqueControl, (others?)
likely zap controlled_kuka directory.
manipulation_station replaces iiwa_world?

RussTedrake, 2020-01-19:
@sammy-tri -- thanks for cleaning up the kuka_examples.  Can you put your specific recommendation for any further action in writing here?

sammy-tri, 2020-01-22:
> likely zap controlled_kuka directory.

Agreed.  It doesn't look like anything from this directory is installed/exported so probably no deprecation period needed.

> manipulation_station replaces iiwa_world?

#12581 started us down this road.

For other stuff which still depends on RBT:

 * kuka_simulation
 * kuka_plan_runner

Should be easy enough to port these and their dependencies away from RBT, I think it's worth doing

 * move_iiwa_ee

Depends on `ConstraintRelaxingIk` which still uses RBT.  I'd like to port it to MBP.  I think this program is a good minimal example of publishing a robot plan and using the plan runner.  If we don't think the "publish plan and have a runner execute it" model is worth having an example of, I think that in the interests of having a simple demo showing drake controlling real hardware, it should get a differential IK version like `move_jaco_ee`.

RussTedrake, 2020-01-27:
Nice.  @siyuanfeng-tri is also working on a robot-bridge #12636 that covers some of these cases, and/or needs the MBT versions of the tools.  @siyuanfeng-tri -- might you weigh in on the right path forward here to be compatible?

jwnimmer-tri, 2022-05-03:
The examples code and such has evolved a bit since this issue was originally posted.  It would help if someone could re-assess the current state of the code, and explain here what the new victory condition would be.

closed 2024-03-20 by jwnimmer-tri