RobotLocomotion/drake #9728: Ghost robot visualization (for planned actions) runs through scenegraph.

RussTedrake, 2018-10-19:
We had/have that workflow when using the full director interface. But it should work for any renderer using scenegraph -- just a matter of registering additional geometry (with reduced alpha).

RussTedrake, 2018-11-11:
Probably depends on support for geometry roles appearing in SceneGraph.  E.g. #9813

RussTedrake, 2020-04-24:
Possibly also depends on parsing directly into SceneGraph, #10775

EricCousineau-TRI, 2020-04-28:
@mposa I think this is y'all's current call-site that could be improved by this feature?
https://github.com/DAIRLab/dairlib/blob/5520e9f5c6345d161eed0c81ce4707e1d459ca9b/multibody/multipose_visualizer.cc#L30-L40

(Ran across it b/c I was looking into your potential ROS usages to possibly mention to @IanTheEngineer)

mposa, 2020-04-28:
Yup, exactly.

RussTedrake, 2022-01-09:
I'm going to close this one.  It's easy to do in meshcat without going through scenegraph  (i do it often in the manipulation notebook now)

closed 2022-01-09 by RussTedrake

RussTedrake, 2023-07-11:
ftr -- here's a possible code snippet which can accomplish this.
```
def DrawRobot(query_object: QueryObject, meshcat_prefix: str, draw_world: bool = True):
    rgba = Rgba(0.7, 0.7, 0.7, 0.3)
    role = Role.kProximity
    # This is a minimal replication of the work done in MeshcatVisualizer.
    inspector = query_object.inspector()
    for frame_id in inspector.GetAllFrameIds():
        if frame_id == inspector.world_frame_id():
            if not draw_world:
                continue
            frame_path = meshcat_prefix
        else:
            frame_path = f"{meshcat_prefix}/{inspector.GetName(frame_id)}"
        frame_path.replace("::", "/")
        frame_has_any_geometry = False
        for geom_id in inspector.GetGeometries(frame_id, role):
            path = f"{frame_path}/{geom_id.get_value()}"
            path.replace("::", "/")
            meshcat.SetObject(path, inspector.GetShape(geom_id), rgba)
            meshcat.SetTransform(path, inspector.GetPoseInFrame(geom_id))
            frame_has_any_geometry = True

        if frame_has_any_geometry:
            X_WF = query_object.GetPoseInWorld(frame_id)
            meshcat.SetTransform(frame_path, X_WF)
```