RobotLocomotion/drake #9732: Replace iiwa14 sdf with iiwa7 urdf/sdf in ManipulationStation

RussTedrake, 2018-10-19:
And/or optionally take the sdf as an argument in the constructor (since we might want to use different collision model versions, etc)

pangtao22, 2018-10-21:
iiwa7_no_collision.sdf can be found here: https://github.com/pangtao22/models/tree/master/iiwa7

RussTedrake, 2018-10-28:
@Sproulx -- does the mounting location of the wsg need to change (in the simulation model) when it's mounted on the iiwa7 instead of iiwa14?  can you confirm?

cc @pangtao22

RussTedrake, 2018-10-28:
also waiting on confirmation of offsets (for visual geometries at least) for iiwa7 models to address the visual errors observed in the visualizers.  @Sproulx was going to check against the original URDF and CAD, then send his recommended updates for both the no-collision model and the box-collision model.

RussTedrake, 2018-10-29:
So the end effector frame in the code would not have been correct for the iiwa14?  (Presumably the other uses of iiwa14 in drake are correct, though?)

Can one of you send me the new end effector weld coordinates for the iiwa7 model?

pangtao22, 2018-10-29:
I don't know how TRI has been mounting the grippers, but the EE frame in the code would not have been correct for the iiwa14 in the lab. 

The correct end effector transformation:
Change the translation in [this line](https://github.com/RussTedrake/drake/blob/d5f89db62ad12975d01674861cd2cd6cd68a07c4/examples/manipulation_station/manipulation_station.cc#L143) to `[0, 0, 0.114]`

Sproulx, 2018-10-29:
@RussTedrake Looks like the 5mm and 6mm offsets I found based on Rick's observation were in the files we used as a starting point to create the SDF and not something that was introduced by us. On closer inspection, just now, I happened to be looking at the rendering of the model without those offsets at just the right angle to notice a gap between the links where the overlap used to be. I corrected that in both the no-collision and box-collision models. I put the two SDFs on dropbox, since GitHub won't let me attach them. 

https://www.dropbox.com/sh/9d3hwu6sq8tihl9/AAATHnnYwlrgz9Wcxp9lq2aha?dl=0

PR #9857 (2018-11-06): Minor fixes to iiwa7 urdf; use iiwa7 in manipulation_station
Resolves #9732

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9857)
commit a7e45b74 (2018-11-06): Minor fixes to iiwa7 urdf; use iiwa7 in manipulation_station

Resolves #9732
closed 2018-11-06 by RussTedrake