RobotLocomotion/drake #9741: update manipulation station setters to put the Context* first,

RussTedrake, 2018-10-20:
In DiscreteDerivative, SchunkWsgPositionController, and StationSimulation, I've used setters that put the Context as the last argument (because it's mutable).  @sherm1 points out "there is a Drake style guide exception specifically for Context -- it should always be the first argument even if mutable."

RussTedrake, 2018-10-21:
this is not done consistently in the code base... even in multibody.  

for instance, I see
```
  void SetFreeBodyPoseOrThrow(
      const Body<T>& body, const Isometry3<T>& X_WB,
      systems::Context<T>* context) const;
```
and friends taking the context last.  

But 
```
  const RevoluteJoint<T>& set_angle(
      Context<T>* context, const T& angle) const {
    get_mobilizer()->set_angle(context, angle);
    return *this;
  }
```

Honestly, I prefer putting the context last.  I think it is more clear semantics, since we are trained everywhere else in the codebase to put mutable pointers last.  Might even say we should revisit that choice in the style guide.

sherm1, 2018-10-21:
There are good reasons for the placement rule, discussed at the time. For example, putting the context at the end would prevent use of default parameters in many cases where they would be useful. Consider this like `this` in C++ or `self` in Python -- basically a meaningless-to-user framework-required argument, best placed at the beginning always and then forgotten.

PR #10318 (2018-12-27): manipulation setters take Context* as the first argument
Resolves #9741

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10318)
commit e0bed1e0 (2018-12-27): manipulation setters take Context* as the first argument

Resolves #9741
closed 2018-12-27 by sherm1