RobotLocomotion/drake #9745: Update documentation style guide: Doxygen style, Python docstrings

EricCousineau-TRI, 2018-10-21:
Things discussed with @jamiesnape:
- [ ] Conventions for Doxygen docstrings; add this to `doc/doxygen_instructions.rst`
  - e.g. spacing on `@code` blocks, where to place `TODO`s, Math styles, etc.
- [ ] Clarify that we use [Napoleon-style docstrings](https://sphinxcontrib-napoleon.readthedocs.io/en/latest/) for Python

EricCousineau-TRI, 2018-10-21:
Re: #9735
@jamiesnape I would agree with @sherm1 on the aesthetics, in that we should permit (or even enforce) indentation in `@code` blocks.
After writing the script to automatically change the indentations, and seeing that Doxygen renders properly with indentations, it should be rather easy to make `mkdoc` behave properly with this behavior, and automatically set the indentations on existing code once we decide on something.

If Doxygen doesn't handle it properly at some point, we can address it then.

jamiesnape, 2018-10-22:
> I would agree with @sherm1 on the aesthetics, in that we should permit (or even enforce) indentation in @code blocks.

We are writing multi-modal documentation and I would not even agree that how documentation looks in source is even the most important modality. But I guess I would be outvoted.

jamiesnape, 2018-10-22:
> Clarify that we use Napoleon-style docstrings for Python

With whom? Our style guide is non-specific, and it is what our tools accept.

jwnimmer-tri, 2018-12-21:
One thing we should do sooner rather than later is to forbid doing this:
```
/// A nice log Doxygen comment.
// An implementation comment.
double CalcSomething(...) { ...; }
```

Our mkdoc tooling loses track of the Doxygen comment when it doesn't directly abut the method.

SeanCurtis-TRI, 2019-01-02:
@mitiguy I don't know if you saw this comment -- but it applies to your style.

mitiguy, 2019-01-08:
@SeanCurtis-TRI  Thanks - message received.  I'll talk with you about best practice, e.g., how to handle comments I formerly labeled with   
// @internal

jwnimmer-tri, 2019-01-08:
@mitiguy For one idea, you can look at what I did at https://github.com/RobotLocomotion/drake/pull/10289/files#diff-1cf56ab04d535b7e08ba6aca2d0c2d31 for `rigid_transform` and `matrix`.  IMO there are probably many fine ways to do it.  Even doing `/// @internal Stuff` is probably fine (adding the third slash).

mitiguy, 2019-01-08:
@jwnimmer-tri -- Thanks for the ideas.  I like the way you did it.

EricCousineau-TRI, 2019-08-01:
Per f2f, should have a first rough-cut PR with just style-guide suggestions (probably no linter to start).

jwnimmer-tri, 2021-11-11:
We have some documentation advice captured in https://drake.mit.edu/styleguide/pyguide.html already now.  What (if anything) still needs to be done to resolve this issue?  What is the victory condition?

closed 2023-01-18 by jwnimmer-tri