RobotLocomotion/drake #9754: mbp: Enable pre-finalization requests for actions that require finalization

EricCousineau-TRI, 2018-10-22:
At present, certain workflows like collision filtering require finalization. While some portions of this workflow (such as getting geometry IDs) are currently marked as needing finalization, but may not actually need it, there are other bits that need topology queries that can realistically only be achieved via finalization (e.g. something like `GetBodiesWeldedTo` in #9740).

Potential solutions:
- [ ] Remove finalization requirements from methods that don't really need it (possibly `GetBodyFrameIdIfExists()`, `GetBodyFrameIdOrThrow()`, etc.)
- [ ] Add some sort of queuing mechanism, either transparently or explicitly.
   - Easiest starting point is explicit queuing.

\cc @sammy-tri @SeanCurtis-TRI

EricCousineau-TRI, 2019-03-29:
Could be done via filter groups; filter groups can be mutated pre-finalize (as they are only SceneGraph). Should create checkbox or another issue for sdf parsing.

amcastro-tri, 2019-04-12:
Even to be **more** specific. We want to get rid of **DRAKE_MBP_THROW_IF_NOT_FINALIZED**:
https://github.com/RobotLocomotion/drake/blob/1b09de88b89a40300ccf047a6af29091c0bd7836/multibody/plant/multibody_plant.cc#L405

PR #11257 (2019-04-25): Welded bodies
Resolves #9754.
MBP now creates a graph of bodies/joints during pre-finalization, which allows users to answer physically meaningful topological questions. In this PR we answer to the question from #9754; what bodies are welded to this one particular body? so that users can create collision filters as they build their model, pre-finalize.

cc'ing @EricCousineau-TRI

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11257)
closed 2019-04-25 by amcastro-tri