RobotLocomotion/drake #9771: Need sugar for connecting rgb-d cameras to drake visualizer

RussTedrake, 2018-10-23:
As discussed with @SeanCurtis-TRI 

.. otherwise we repeat the history that got us to ConnectDrakeVisualizer.

I have a working draft in `systems/sensors/visualization/` in RussTedrake/drake.git branch lab1.  Will shake it out over the next few hours, then potentially hand it off to @SeanCurtis-TRI to clean up and PR.

SeanCurtis-TRI, 2019-09-10:
@RussTedrake in revisiting this issue, I'm left wondering what the actual statement of work is. The title suggests something similar to `ConnectDrakeVisualizer`, but the body seems to be holding that up as a warning. There was never a hand-off *per se*.

Can we revisit what the feature being requested here is so that it becomes actionable?

RussTedrake, 2019-09-11:
Sorry for the poor write-up.  I think it's safe to close this unless I can find something more concrete.

closed 2019-09-11 by RussTedrake