RobotLocomotion/drake #9779: geometry: truncated signed-distance field from all objects in a frame.

DamrongGuoy, 2018-10-23:
We need the ability to query the signed-distance field from each sphere in each frame to all objects in a specific frame.  We assume all query spheres locate within a user-defined bounding box.  In other words, the signed-distance field is truncated within the bounding box.  The accuracy of the result is controlled by the cell-size parameter that controls the resolution of the underlying Cartesian grid cells (voxels) of the user-defined bounding box.

The query will be available in `drake::geometry::QueryObject`.  The signed distance field will be initialized and stored in `drake::geometry::SceneGraph`.

We will use `voxelized_geometry_tools` and `common_robotics_utilities` for the core functionality (`github.com/calderpg/voxelized_geometry_tools` and `github.com/calderpg/common_robotics_utilities`).

We will continue to update the problem description above and the API below as we discuss this issue together.

```
drake/geometry/scene_graph.h
/**
  Registers and initializes signed distance field.
  @param source_id  The id of the source registering the signed distance field.
  @param frame_id   The id of the frame on which the signed distance field is
                    defined. All objects fixed to this frame collectively
                    define the signed distance field.
  @param min_point  The corner point with minimum coordinates of the bounding
                    box on which the signed distance field is defined.
  @param max_point  The corner point with maximum coordinates of the bounding
                    box on which the signed distance field is defined.
  @param cell_size  Cell size of the voxels that subdivide the bounding box on
                    which the signed distance field is defined.  The smaller
                    the cell size is, the more accurate the signed distance
                    field becomes.
  @return  The id of the signed distance field.
*/
SdFieldId RegisterSignedDistanceField(
    SourceId source_id,
    FrameId frame_id,
    const Vector3<double>& min_point,
    const Vector3<double>& mmax_point,
    double cell_size);
```
```
drake/geometry/query_object.h
/**
  Query the signed distance field at every sphere in the geometry source.
  @param signed_distance_id  The id of the signed distance field that was
                            registered and initialized in the SceneGraph.
  @return  A list of signed-distance field value (see SdFieldValue), each of which
           corresponds to a sphere in a frame in the source.
*/
std::vector<SdFieldValue<T>> QuerySignedDistanceField(
    SdFieldId signed_distance_id);

/**
  Signed-distance field data.  For a query sphere corresponding to `sphere_id`,
  the geometry `anchored_id` corresponds to the closest anchored geometry in
  the frame on which the signed distance field is defined. The `distance` is
  the distance between the sphere and the anchored geometry.  The `gradient`
  is the gradient of the signed distance field at the center of the sphere.
  The `nearest_point` is the point on the anchored geometry closest to the
  sphere.
*/
template <typename T>
struct SdFieldValue {
  GeometryId sphere_id;
  GeometryId anchored_id;
  T  distance;
  Vector3<T> gradient;
  Vector3<T> nearest_point;
};
```

DamrongGuoy, 2018-10-23:
@hongkai-dai, @calderpg, @thduynguyen, @sherm1, @SeanCurtis-TRI for comments.

sherm1, 2018-10-24:
A few quick comments:
- Specifying the box size and cell size independently can produce a non-integer number of cells. Consider box size + number of cells or some other combination that works. Otherwise need to say what happens (box is grown a little? cells shrunk? error message?)
- The names here seem too generic. Clearly this interface is tied very tightly to a particular narrow query that is implemented using a non-adaptive grid and queried only with spheres, and only against anchored geometry. I would prefer a more-general interface that isn't so closely tied to the underlying implementation, but if we're going to make one like this the names need more adjectives.
- This is a somewhat-awkward way to discuss the API since we can't comment on individual lines. Consider a Google doc instead (could link from here).

tri-ltyyu, 2018-10-24:
Action items from meeting on 10/24:
-> Damrong will update API based on requirements from Hongkai, Duy and Calder. We'll start with FCL and update with faster implementation later to fulfill Duy and Calder's needs
-> Calder and Duy will meet to discuss voxel space requirements

sherm1, 2018-10-25:
Immediate action plan (from further discussion 10/24): 

Damrong is going to work up a test program to put in Drake master that 
- builds an MBP-SG Diagram containing an assortment of geometry
- sets q's to position the geometry
- queries SG with a point "probe" P that returns 
    - signed distance d from P to the closest geometry G
    - G's id and the nearest point Gn on G
    - the gradient ∂d/∂P

Note that this is _not_ the API described above; this is just to make sure the basics are in place. We hope to have this at least in sketchy form next week.

- variants for after that works
    - return above info for distances from P to _all_ geometries
    - same but only for geometries within a certain radius of P, or selected geometry subsets
    - handle a batch of point queries at once and return matrix results
    - make any needed API modifications (if any) for sphere probes rather than points

The idea with this test program is to expose holes where our intended workflow breaks and plug them with minimalist implementations. Once the workflow is in place we can measure performance and accuracy to bring them up to the levels needed by Hongkai, Calder, and Duy.

Discussion and refinements for this plan welcome.

sherm1, 2018-10-29:
Per discussion with @hongkai-dai, @tri-ltyyu, @calderpg-tri, it would be useful to have an early spiral of distance queries even with very limited geometry, as long as it is high precision to provide meaningful gradients. 
- sphere-sphere allows for much of IK to be written and debugged, and (possibly) would allow progress in planning
- @calderpg-tri says adding point-box would be very helpful.

calderpg, 2018-10-29:
From my understanding, what IK needs at minimum is point->sphere(s).

The bare minimum that we need for planning is point->sphere(s) and point->box(es). Our SDFormat environments are entirely collision boxes. It's unclear to me why point->box queries are significantly more complicated than point->sphere - is the perceived bottleneck here in SceneGraph or FCL?

sherm1, 2018-10-29:
Boxes have sharp corners and the signed distance field has discontinuous gradients. We'll need to treat that more carefully and with additional testing -- but sounds like joint point/sphere (or sphere/sphere) isn't enough to get you anywhere.

calderpg, 2018-10-30:
I've been working on some cleanup of the collision checking interfaces in anzu, and these are relevant to the discussion here. See anzu PR [#1369](https://github.awsinternal.tri.global/robotics/enabling-anzu/pull/1369), in particular `ComputePointSignedDistanceAndGradient` in `sphere_robot_model_collision_checker.h` which wraps all the distance and gradient queries needed by the collision checker:

```
PointSignedDistanceAndGradientResult
ComputePointSignedDistanceAndGradient(const Eigen::Vector4d& p_WQ, 
                                      double query_radius) const
```

where `PointSignedDistanceAndGradientResult` wraps the minimum signed distance and (potentially) multiple (signed distance, gradient) pairs. This allows the caller to handle combining multiple gradients however they want to. `query_radius` is used to limit the number of distance and gradients that need to be computed. For distance queries against SceneGraph, this would ideally provide distances and gradients for all objects within `query_radius`.

From an overall perspective, `MBPEnvironmentCollisionChecker`'s implementation of `ComputePointSignedDistanceAndGradient` is the missing link for a first implementation of a planning collision checker that uses MBP + SG in anzu.

DamrongGuoy, 2018-10-31:
Thanks a lot! @calderpg 
That is very useful.

SeanCurtis-TRI, 2018-11-01:
I have a different perspective that makes me a bit worried:

Assuming that your picture implies that the manipulator is represented by a bunch of sphers.

1. Interpretation 1: the query points arise from manipulator geometry. You'd get distance/gradient between the top orange circle and bottom orange circle (assuming that each belongs to a different body, non-adjacent body). This is based on the default collision filtering due to MBP.,
2. Interpretation 2: The API that has been proposed does *not* correlate query points with bodies, frames, or geometries. That means there is no filtering (beyond the maximum distance). So, from this view, you'll get a distances from a query point to *every* geometry (including the manipulator geometry).

If either of these interpretations are wrong, someone should correct me. Otherwise, people should be aware of this and either buy into it or express concern.

sherm1, 2018-11-01:
Thanks for that great diagram, @EricCousineau-TRI !

Per Sean's comment, I wonder if the query point will need to be accompanied by a filtering specification (maybe as an option after we do the non-filtered one) ?

sherm1, 2018-11-01:
Would it make sense to probe with a point "as if" it belonged to some particular piece of geometry for which geometry set inclusions have already been established?

SeanCurtis-TRI, 2018-11-01:
That's another mechanism -- you could associate it with a geometry id and inherit that geometry's filter logic.

DamrongGuoy, 2019-01-08:
Last week we landed PR #10148.  Now Drake support signed distances from all spheres and boxes to a query point.  Now I assign [anzu issue #1275](https://github.awsinternal.tri.global/robotics/enabling-anzu/issues/1275) to @calderpg for testing.  For reference, these are the relevant doxygen pages.
```
drake::geometry::QueryObject::ComputeSignedDistanceToPoint();
drake::geometry::SignedDistanceToPoint;
```
[image]
[image]

DamrongGuoy, 2019-01-08:
As a side note, we also have a relevant issue #10270 to calculate signed distances between spheres and spheres accurately. Its solution PR #10281 shares code with the above solution and will land soon, hopefully.