RobotLocomotion/drake #9781: Determine if multibody/parsing is dead code

amcastro-tri, 2018-10-23:
The code in `multibody/parsing` was initially written to get the ball rolling for the interaction with OSRC to help us with the SDF parser. The decision was to migrate features from that code directly into `sdformat` and it slowly decayed into disuse.

If nothing from this directory is useful (for instance the `frame_cache.h` code used to specify the pose of a frame in any other frame defined int the file), this code should be eliminated from Drake master.

@sammy-tri, @jwnimmer-tri, anything useful there you'd like to keep?

jwnimmer-tri, 2018-10-23:
We can always go back and resurrect (in a new home) anything we end up wanting.  I'm okay to remove it.

jwnimmer-tri, 2018-10-24:
I opened a PR.  Sam can bless the idea there.

PR #9800 (2018-10-25): multibody/parsing: Remove entire package of dead code
Closes #9781.  Relates #9314.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9800)
closed 2018-10-25 by jwnimmer-tri