RobotLocomotion/drake #9782: Ospray renderer cannot be used without OpenGL context

SeanCurtis-TRI, 2018-10-23:
## Description

The OSPRay bindings in VTK are billed as ["a render pass that uses OSPRay instead of GL"](https://www.vtk.org/doc/nightly/html/classvtkOSPRayPass.html#details) and OSPRay stems from Intel. As such, it is not a library or algorithm that depends on a graphics card. *However*, the nature of the OSPRay bindings are such that if no OpenGL context can be created, the OSPRay renderer cannot be run. One of the implications is that if a simulation has an RgbdCamera using an OSPRay-based renderer, it cannot be run "offline" (e.g., via ssh on a remote computer).

An [issue](https://gitlab.kitware.com/vtk/vtk/issues/17422) has been posted with VTK and Drake's OSPRay based rendering should be updated as necessary when offline rendering is possible.

jamiesnape, 2019-02-26:
Do you want to assign this to me since it needs VTK changes?

SeanCurtis-TRI, 2019-02-26:
You're welcome to it.  :)  I'm going to keep my name affixed to it as well for bookkeeping, but I appreciate you tackling this.

jwnimmer-tri, 2021-07-24:
Obsoleted via #14339.

closed 2021-07-24 by jwnimmer-tri