RobotLocomotion/drake #9783: Disposition of files that depend on //attic

jwnimmer-tri, 2018-10-23:
The following non-attic packages have dependencies into the attic.  Ideally, nothing outside the attic should depend inside the attic.

```console
$ bazel query '
  let foo = (//...
    except attr(tags, lint, //...)
    except //attic/...
    except //bindings/...
    except //tools/...) in
  $foo intersect allpaths($foo, //attic/...)' \
  --output=package

examples
examples/acrobot
examples/allegro_hand
examples/compass_gait
examples/contact_model
examples/double_pendulum
examples/humanoid_controller
examples/humanoid_controller/dev
examples/kinova_jaco_arm
examples/kuka_iiwa_arm
examples/kuka_iiwa_arm/controlled_kuka
examples/kuka_iiwa_arm/dev/box_rotation
examples/kuka_iiwa_arm/dev/monolithic_pick_and_place
examples/kuka_iiwa_arm/dev/pick_and_place
examples/kuka_iiwa_arm/iiwa_world
examples/kuka_iiwa_arm/pick_and_place
examples/manipulation_station
examples/particles
examples/pendulum
examples/pr2
examples/qp_inverse_dynamics
examples/quadrotor
examples/rimless_wheel
examples/rod2d
examples/schunk_wsg
examples/valkyrie
manipulation/dev
manipulation/models/allegro_hand_description
manipulation/models/iiwa_description
manipulation/models/jaco_description
manipulation/models/wsg_50_description
manipulation/planner
manipulation/scene_generation
manipulation/schunk_wsg
manipulation/sensors
manipulation/util
multibody
multibody/benchmarks/free_body
multibody/collision
multibody/joints
multibody/parsers
multibody/rigid_body_plant
multibody/shapes
perception
perception/estimators/dev
systems/analysis
systems/controllers
systems/controllers/plan_eval
systems/controllers/qp_inverse_dynamics
systems/controllers/test_utilities
systems/estimators/dev
systems/rendering
systems/robotInterfaces
systems/sensors
systems/trajectory_optimization
```

(Note that a couple packages in the above are false positives, due to the install rules.)

We should formulate a policy & plan here, and then PR it.

Relates #9343.

jwnimmer-tri, 2018-10-23:
My proposal is:
(1) We leave all `//examples//...` alone.  Those can depend on the attic, and can be ported incrementally.  They are not reusable library code, but rather just sometimes-installed demos.  It's okay for them to reach into the attic, if they are an attic demo.
(2) Everything else that (transitively) depends on RBT, we just move into the attic -- we don't try to upgrade it in place.  If someone wants to upgrade it to MBP later, they can move it out of the attic.

SeanCurtis-TRI, 2018-10-23:
I'd like to elaborate the strategy for resolving things in the attic (specifically those being upgraded to MBP). For example, how do we handle a class that would then have two different forms: attic-compatible and non-attic compatible. The class I have currently in mind to fuel the question is `RgbdCamera`.

In its current incarnation, it is wholly dependent on RBT (attic contents). In its new incarnation, it does not. However, I can't have two classes with the same name in attic/non-attic parallel directories due to the games the attic plays with install (as I understand them). So, that seems to leave us two options:

1. Kill the attic version entirely during the update (losing the attic-compatible version), or
2. Use different names.

Other options? Suggestions? Corrections?

jamiesnape, 2018-10-24:
I think at the stage you need to start reusing names, maybe things need to start getting more aggressive and have `deprecated` namespaces and `deprecated` folders as part of the include path. Causes a little pain downstream, but eventually you need to get to a point where you are on the verge of removing something. We have been fairly kind to people so far, but if you give a fair warning to people, (maybe) they will understand.

Of course punting and using a different name is a little friendlier...

jwnimmer-tri, 2018-10-24:
I've updated the original post to list the exact targets, instead of just the packages, and to omit the false positives and such.

@RussTedrake are the any components listed above, that you would _not_ want to see go into the attic?

RussTedrake, 2018-10-25:
That's an extremely aggressive culling of the code.

`trajectory_optimization` should NOT go to the attic.  the target you list above only has the dependency because it is the roll-up target with the `_evaluator` classes.  I'm more ok with those moving to the attic.

The inverse dynamics / inverse dynamics controller have been upgraded to support both RBT and MBP (with different constructors on the same class).  That recipe is clearly at odds with your approach here.  We do not want to attic the inverse dynamics MBP controller.  It is being (heavily) used.  same is true for differential inverse kinematics.  maybe more.

I'm not convinced that this has been thought out carefully enough to move on it yet.
can you cross-reference your list with the targets that ALSO depend on MBP/SceneGraph?

jwnimmer-tri, 2018-10-25:
The rollups like `//systems/trajectory_optimization:trajectory_optimization` would of course stay, after their RBT-only components are moved.  I know that some classes like ID support both, and we'd also not attic those.

I guess I was looking for any simple advice before I start digging into the details of each case one by one, but maybe that's not practical.  Instead, I'll just work through the details and open a PR, and go from there.  That way it'll be specific and well thought-through.

RussTedrake, 2018-10-25:
Sounds good.  Thanks.  I am ok with moving large chunks of the code to force the transition.