RobotLocomotion/drake #9791: unordered_set erase issue causes assertion in geometry dev

RussTedrake, 2018-10-24:
The error message is
```
abort: Failure at geometry/dev/geometry_state.cc:257 in operator=(): condition 'initial_size > unprocessed.size()' failed.
```

I believe the problem is clear:  you are doing a range-based loop and performing an erase inside the loop.  You're not allowed to do that.  (c.f. https://stackoverflow.com/a/20317088/9510020)

What I don't have time to understand, yet, is why it manifests itself in my python code, but not my c++ tests.  But it happens in my RussTedrake/drake.git lab1 branch when I `bazel run //examples/manipulation_station:joint_teleop`.  

unfortunately, this one is on fire.  Can you help me with a quick fix?

RussTedrake, 2018-10-24:
cc @pangtao22

SeanCurtis-TRI, 2018-10-30:
@RussTedrake this should probably closed due to PR #9796. Do you concur?

closed 2018-10-30 by RussTedrake