RobotLocomotion/drake #9802: mac: Experimental `*-cache` build may hang on linking of `runge_kutta3_integrator`?

EricCousineau-TRI, 2018-10-25:
See comment here:
https://github.com/RobotLocomotion/drake/pull/9768#issuecomment-432893245

From this build: https://drake-jenkins.csail.mit.edu/job/mac-highsierra-clang-bazel-experimental-cache/15/
```
21:46:21 [15,295 / 15,347] 1071 / 4129 tests; Linking systems/analysis/librunge_kutta3_integrator.a [for host]; 1730s remote-cache ... (3 actions running)
21:46:21 INFO: From Linking examples/pr2/pr2_passive_simulation:
...
21:51:06 INFO: From Linking examples/quadrotor/run_quadrotor_dynamics_test:
21:51:06 clang: warning: argument unused during compilation: '-pie' [-Wunused-command-line-argument]
21:52:55 [25,250 / 25,299] 4128 / 4129 tests; Linking systems/analysis/librunge_kutta3_integrator.a [for host]; 2123s remote-cache
22:00:58 [25,250 / 25,299] 4128 / 4129 tests; Linking systems/analysis/librunge_kutta3_integrator.a [for host]; 2607s remote-cache
22:10:02 [25,250 / 25,299] 4128 / 4129 tests; Linking systems/analysis/librunge_kutta3_integrator.a [for host]; 3151s remote-cache
22:20:06 [25,250 / 25,299] 4128 / 4129 tests; Linking systems/analysis/librunge_kutta3_integrator.a [for host]; 3755s remote-cache
22:31:55 [25,250 / 25,299] 4128 / 4129 tests; Linking systems/analysis/librunge_kutta3_integrator.a [for host]; 4464s remote-cache
22:45:16 [25,250 / 25,299] 4128 / 4129 tests; Linking systems/analysis/librunge_kutta3_integrator.a [for host]; 5264s remote-cache
23:01:20 [25,250 / 25,299] 4128 / 4129 tests; Linking systems/analysis/librunge_kutta3_integrator.a [for host]; 6229s remote-cache
```

EricCousineau-TRI, 2019-11-19:
Marking this closed. Haven't seen it in a while?

closed 2019-11-19 by EricCousineau-TRI

jamiesnape, 2019-11-19:
We have not had cached Mac builds for quite some time, so we would not see it...