RobotLocomotion/drake #9803: Upload SNOPT 7.4 sources into git for use by CI

jwnimmer-tri, 2018-10-25:
As it turns out, every version of SNOPT has various deficiencies, with no single version being monotonically superior to all others.  In order to support ongoing development of MathematicalProgram, we're going to want Drake CI to have access to several revisions of SNOPT, as we explore exactly how to mitigate the deficiencies.

To that end, we'd like to ask that MIT (i.e., @RussTedrake) obtain SNOPT version 7.4 source for use by Drake CI, and that the CI maintainers (i.e., @jamiesnape) uploaded that source to [RobotLocomotion/snopt](https://github.com/RobotLocomotion/snopt/) using some tag like we did with SNOPT 7.6.

(For the record, the SNOPT 7.6 source upload was tracked in #8060.)

Marked as high priority because it relates to #4893 (running SNOPT in parallel), which is a current TRI priority.

jwnimmer-tri, 2018-10-25:
@RussTedrake I forwarded you emails of TRI's SNOPT 7.4 discussions with SBSI, to kick this off.

jwnimmer-tri, 2018-10-25:
\CC @EricCousineau-TRI @hongkai-dai @avalenzu @calderpg-tri @sammy-tri

EricCousineau-TRI, 2018-12-06:
@RussTedrake Can I ask if you've had a chance to reach out to SBSI for the sources? Is there anything I can help with?

tri-ltyyu, 2019-01-03:
Sent follow-up to SBSI sales

RussTedrake, 2019-01-03:
sorry that this conversation moved offline.  i forwarded the source code from the snopt developers to @jamiesnape about a month ago.

jamiesnape, 2019-01-03:
I am not seeing that email. Can you forward it again?

jwnimmer-tri, 2019-01-03:
The source release for SNOPT 7.4-1.1 is now available at `RobotLocomotion/snopt` branch `2015-02-26` commit `0254e961cb8c60193b0862a0428fd6a42bfb5243`.

closed 2019-01-03 by jwnimmer-tri